Kinovarobotics/kinova-ros #253: publish full finger states

christian-rauch, 2019-07-29:
## Description

The finger states are published on topic `out/finger_position` as angles (degree). These angles are converted by factor `1/6800*80*M_PI/180` (0.000205333) to joint state values:
https://github.com/Kinovarobotics/kinova-ros/blob/7c0fc4d28ac04f072884b1373661fe80dce9b35b/kinova_driver/src/kinova_arm.cpp#L549-L551
The `robot_state_publisher` therefore publishes finger transforms (for the proximal fingers) with a very small movement.
The alternative `kinova_tf_updater` does not handle finger states at all.

The distal fingers are underactuated and do not change their transform (with respect to the proximal finger) if the fingers are not in contact with an object.

## Version

ROS distribution : melodic

Branch and commit you are using : 

## Steps to reproduce

1. run the real robot, open & close gripper
2. log the proximal finger transform: `rosrun tf tf_echo /j2n6s300_link_6 /j2n6s300_link_finger_2`
3. recognise that the transformation does not change

## Expected behavior

The proper finger states should be published as joint states for FK.

alexvannobel, 2019-08-09:
See #254 for the discussion on how to resolve this issue.

alexvannobel, 2019-08-09:
@christian-rauch feel free to reopen this issue if something's wrong with my changes!

PR #259 (2019-08-09): Added a parameter in the driver to visualize correctly the Jaco 1 fingers
Fixes #253 .
This adds the ```use_jaco_v1_fingers``` parameter to the kinova_driver node because the Jaco 1 fingers don't have position values in the same range as the Jaco 2 fingers and this was not dealt with in the driver.
closed 2019-08-09 by alexvannobel