Kinovarobotics/kinova-ros #258: Unclear moveit documentation for ros-kinetic

Hubert51, 2019-08-08:
## Description
I am trying to follow the [wiki](https://github.com/Kinovarobotics/kinova-ros/wiki/MoveIt) to use moveit to command the arm. After run these two lines in the terminal, I try to move the arm in rviz.  
In first terminal
`roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=*robot_type*`  
In second terminal  
`roslaunch robot_name_moveit_config robot_name_demo.launch`  
But the arm is not moving after I set the goal and execute it. The output in second terminal is 
```
[ INFO] [1565293170.497931932]: SimpleSetup: Path simplification took 0.004246 seconds and changed from 3 to 2 states
[ INFO] [1565293170.575530872]: Joint_trajectory_action_server received goal!
[ INFO] [1565293170.575621877]: Joint_trajectory_action_server accepted goal!
[ INFO] [1565293170.575662005]: Joint_trajectory_action_server published goal via command publisher!
[ INFO] [1565293171.674251140]: Completed trajectory execution with status SUCCEEDED ..
```

## Version

ROS distribution : kinetic in ubuntu16.04

Branch and commit you are using : kinetic devel

## Suggestion
1. Write the documentation better
In ros-kinetic branch, I found we need to run three scripts to use moveit
```
roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=*robot_type*  
roslaunch robot_name_moveit_config robot_name_demo.launch  
rosrun kinova_driver pid_traj_action_server.py
```
2. The `pid_traj_action_server.py` code is specific for `j2s7s300` arm. Can you write code to let the code fit all the arm

alexvannobel, 2019-08-08:
Hi @Hubert51 ,
First off, can you try the master branch? There were a couple recent updates to make it work with Kinetic. With this branch, I can send MoveIt plans with RViz after only two commands : 
First start the driver: 
```roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=j2s7s300```
and then start MoveIt and RViz :
```roslaunch j2s7s300_moveit_config j2s7s300_demo.launch```.

Do those steps work for you?

About the documentation, I am aware that it is lacking and not up-to-date and we're looking at cleaning this repository and making it much clearer in the coming weeks, but I'll address the documentation issue on its own. 

Hope this helps,
Cheers,
Alex

Hubert51, 2019-08-11:
Yeah, I can try! But if I want to use j2n6s300, I think I can not use kinetic branch. Is it correct?

alexvannobel, 2019-08-12:
Hi @Hubert51 
I just tried it with ROS Kinetic (master branch) and I was able to move the robot with MoveIt.
Do you have error messages when you launch the nodes?

Hubert51, 2019-08-12:
I use kinetic branch. In kinetic branch, I can not move the j2n6s300 robot with Moveit. I did not try it with master branch. Can you try kinetic branch code. I think the code can not move the j2n6s300 arm.

Hubert51, 2019-08-12:
Ok, I tried code from my partner, which I did not know the branch. Then I can use `roslaunchj2n6s300_moveit_config j2n6s300_demo.launch` to run the arm with moveit. I think this code should be from master branch. So master branch code should be good. But code in kinetic branch may be not good

alexvannobel, 2019-08-12:
I cleaned many stale branches in the repo recently, so now the branch to use is **master** if you are on Kinetic or Melodic (I removed the other ones). There is still a **indigo-devel** branch, but this one won't be maintained in the future. 
Glad your problem is solved!

closed 2019-08-12 by alexvannobel