Kinovarobotics/kinova-ros #261: The wildcard command no longer works

mpatalberta, 2019-08-14:
## Description
roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=*robot_type*
No longer finds a robot
Ros
Melodic

Branch
Master
Aug 8 2019 pulled tip of master

## Code example (if necessary)

```
    # paste code here #
```

## Expected behavior

Describe the expected behavior (in regard to the bug described above).

## Any other information

Any other information you believe the maintainers need to know.

alexvannobel, 2019-08-14:
Hi @mpatalberta ,

On ROS Melodic, with the latest `master` branch commit, if I run : 
```roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=j2n6s300```
with a Jaco2 connected to my laptop, it works and the driver initializes.

- What is the exact command you are running? 
- Is your arm properly connected and powered on? 
- What's your console output when you run the command?

mpatalberta, 2019-09-04:
Misundertstandingof the documentation. This is not a problem.

closed 2019-09-04 by mpatalberta