Kinovarobotics/kinova-ros #263: some problems about model predictive control based path following on Jaco2

xuhuairuogu, 2019-08-28:
Hi, I have finished MPC based path following simulation in V-rep and am considering realizing it on the real Jaco2. Now, I have some problems to be solved:
1. How to get dynamical parameters(mass, COM, and inertia of links and actuators)
   In simulation, I used the dynamic model of Jaco2 provided by V-rep. I don't know if the simulated Jaco2 and the real share the same dynamic parameters. By the way, I haven't found any documents about dynamic parameters of Jaco2 on the Kinova website. So, the only way for me is to do dynamic parameters identification. And another two problem arises:
2. How to send trajectory points of joint position to Jaco2 periodically
  In simulation, I adopted the 5-order Fourier form as the excitation trajectory and send joint position commands at the rate of 100HZ.
3. What is the feedback frequency of Jaco2 states(joint position, velocity and torque)
 In simulation, the states feedback frequency is the same as that of position control
 The last one problem is how to do joint torque control on Jaco2.
 Looking forward to your answer!

alexvannobel, 2019-09-05:
Hi @xuhuairuogu ,

1. I am not familiar with the V-rep simulation for Jaco 2, so I don't know how it was obtained, but it may be worth trying. The parameters in [the itertia XACRO file](https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_description/urdf/kinova_inertial.xacro) can also be a good starting point, but the links are approximated as cylinders. I can do a little searching to see if we have better values to provide you.
2. The driver exposes an action server to send Joint Trajectories through the ```/your_robot_name_driver/joints_action/joint_angles/``` action, and it does not provide an interface to send joint positions periodically. However, you can use the ```/your_robot_name_driver/in/joint_velocity``` topic to send joint velocities periodically to the robot. Jaco2 accepts API calls at the maximum rate of 100Hz, so I don't think you'll get to 100Hz joint velocity control because the driver also sends periodic calls to get feedback from the robot. You can try 100Hz at first, but the arm will most likely not catch all the velocity commands. Probably lowering it down to 50 or 60 Hz will give better results (no velocity command will be lost).
3. The ```status_interval_seconds``` parameter is used in the driver to set the frequency for feedback. The default value is 0.1, so 10 Hz. 

Do you also want to do torque control from ROS with Jaco2?

Hope this helps,
Best,
Alex

xuhuairuogu, 2019-09-05:
Thanks for your reply! The ultimate goal is to do torque control based on model predictive strategy. The position or velocity control is only used to make jaco2 follow the optimal excitation trajectory for dynamic parameters identification. By Ethernet connection, it seems that jaco2 could follow trajectory with velocity commands at 100HZ and joint states are feedback at 100HZ at the same time.

closed 2021-06-02 by felixmaisonneuve