Kinovarobotics/kinova-ros #265: How to use kinova in 18.04 Ubuntu？

BlakeQuan, 2019-09-02:
version：18.04ubuntu

When I try to compile, it prompts errors：

 error: no matching function for call to ‘boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000>::subsecond_duration(float)’

mpatalberta, 2019-09-04:
What version of ros are you using. I am running with ubuntu 18.04 with melodic.
I am using the masterbranch of kinova

BlakeQuan, 2019-09-05:
> 你使用的是什么版本的ros。我正在使用带有旋律的ubuntu 18.04。
> 我正在使用kinova的masterbranch

I am using the jaco 2-6 of kinova and running with ubuntu 18.04 with melodic. Besides,I'm a beginner.

alexvannobel, 2019-09-05:
Hi @BlakeQuan ,

Can you make sure you checked out the ```master``` branch? It works with Melodic. 
Apart from that, the error you have is a bit weird. Is your ROS Melodic installation correct?

Best,
Alex

BlakeQuan, 2019-12-01:
After the update, the issue has been resolved

closed 2019-12-01 by BlakeQuan