Kinovarobotics/kinova-ros #268: How to change joystick control

kurenaloha, 2019-09-25:
I want to change the value that is actually input to the robot from the value that is input with the joystick.

I will send  the value of cartesian goals  to the robot action server.
What topic is the joystick value publised in?

closed 2019-09-26 by kurenaloha

rsaif22, 2023-05-10:
Hi, did you find the answer to this?

felixmaisonneuve, 2023-05-10:
Hi @rsaif22 

Joystick Value can be sent by the [SendJoystickCommand](https://github.com/Kinovarobotics/kinova-ros/blob/melodic-devel/kinova_driver/include/kinova/Kinova.API.USBCommandLayerUbuntu.h#L161) API function. 

It is currently not bind to any subscriber, but you can always change your kinova_arm.cpp driver to create your own topic and use it after

Best,
Felix