Kinovarobotics/kinova-ros #269: Cartesian velocity fails after finger action (sometimes)

mfry90, 2019-09-26:
## Description

There are times when, after issuing open/close gripper commands, I no longer have cartesian velocity control.  This doesn't always happen, but it typically happen when the grippers are near the floor.  Curiously, I still have joint velocity control.

## Version

ROS Kinetic
kinova-ros branch/commit: master f89af95 (originally observed in a very old branch: kinetic-devel)

## Steps to reproduce

My node publishes to /j2n6s300_driver/in/cartesian_velocity and /j2n6s300_driver/in/joint_velocity, and it sends goals via an actionlib::SimpleActionClient<kinova_msgs::SetFingersPositionAction> client. I'd rather not post a MWE, but you can reproduce the same problem with the following

Step 1: roslaunch kinova_bringup kinova_robot.launch

Step 2: POSITION THE GRIPPER NEAR THE GROUND (not always necessary, but the failure often occurs in this position)

Step 3: rostopic pub -r 100 /j2n6s300_driver/in/cartesian_velocity kiva_msgs/PoseVelocity "{twist_linear_x: 1.0, twist_linear_y: 0.0, twist_linear_z: 0.0, twist_angular_x: 0.0, twist_angular_y: 0.0, twist_angular_z: 0.0}" 

Step 4: rosrun kinova_demo fingers_action_client.py j2n6s300 percent -- 100 100 100

Step 5: rostopic pub -r 100 /j2n6s300_driver/in/cartesian_velocity kiva_msgs/PoseVelocity "{twist_linear_x: 1.0, twist_linear_y: 0.0, twist_linear_z: 0.0, twist_angular_x: 0.0, twist_angular_y: 0.0, twist_angular_z: 0.0}" 

Step 6: rostopic pub -r 100 /j2n6s300_driver/in/joint_velocity kinova_msgs/JointVelocity "{joint1: 1.0, joint2: 0.0, joint3: 0.0, joint4: 0.0, joint5: 0.0, joint6: 1.0}"

When publishing to cartesian_velocity the 2nd time (Step 5), the arm won't move, but it will when publishing to joint_velocity (Step 6).  This failure doesn't always occur, but oftentimes when the gripper is near the ground (AKA, right after picking up the object of interest!).  

I can regain cartesian velocity control by briefly holding the yellow "return to home" button on the Jaco joystick.

This is the output after closing the grippers ("rosrun kinova_demo fingers_action_client.py j2n6s300 percent -- 100 100 100" from the previous sequence of commands):
[DEBUG] [1569516663.735695528]: The action server has received a new goal request
[DEBUG] [1569516663.735852497]: A new goal has been recieved by the single goal action server
[DEBUG] [1569516663.736221612]: Accepting a new goal
[DEBUG] [1569516663.736333862]: Accepting goal, id: /j2n6s300_gripper_workout-1-1569516663.734, stamp: 1569516663.73
[DEBUG] [1569516665.066863514]: Setting the current goal as succeeded
[DEBUG] [1569516665.067031367]: Setting status to succeeded on goal, id: /j2n6s300_gripper_workout-1-1569516663.734, stamp: 1569516663.73
[DEBUG] [1569516665.067145900]: Publishing result for goal with id: /j2n6s300_gripper_workout-1-1569516663.734 and stamp: 1569516663.73
[DEBUG] [1569516665.067423259]: void kinova::KinovaFingersActionServer::actionCallback(const SetFingersPositionGoalConstPtr&): LINE 141, setSucceeded

mfry90, 2019-10-24:
I've come to understand that the issue originates from the driver switching to angular control instead of remaining in cartesian control.  I believe this is because the end effector is near a singularity position (https://github.com/Kinovarobotics/kinova-ros/issues/113).  Though the @martine1406 changed kinova_comm.cpp to remain in Cartesian control when using the fingers, the solution isn't working for me.  After getting into particular positions and actuating fingers, even with the fix from @martine1406, I still remain in Angular control mode (for example, after I lose Cartesian control, when I move the joystick it sends Joint, rather than Cartesian, commands). 

I'm also not able to reset the node to Cartesian control when I'm in the "freezing" positions (I tried kinova_api_.setCartesianControl, and it doesn't actually change the control to Cartesian).

To replicate how I can't get out of Angular control, run the following commands:

`
roslaunch kinova_bringup kinova_robot.launch

#Moves fine in cartesian:
rostopic pub -r 100 /j2n6s300_driver/in/cartesian_velocity kinova_msgs/PoseVelocity "{twist_linear_x: 1.0, twist_linear_y: 0.0, twist_linear_z: 0.0, twist_angular_x: 0.0, twist_angular_y: 0.0, twist_angular_z: 0.0}" 

#Put into Cartesian "freezing" position:
rosrun kinova_demo joints_action_client.py -v j2n6s300 degree -- 178 270 109 -4.3 32 437

#Won't move in cartesian:
rostopic pub -r 100 /j2n6s300_driver/in/cartesian_velocity kinova_msgs/PoseVelocity "{twist_linear_x: 1.0, twist_linear_y: 0.0, twist_linear_z: 0.0, twist_angular_x: 0.0, twist_angular_y: 0.0, twist_angular_z: 0.0}" 

#Will move in angular:
rostopic pub -r 100 /j2n6s300_driver/in/joint_velocity kinova_msgs/JointVelocity "{joint1: 1.0, joint2: 0.0, joint3: 0.0, joint4: 0.0, joint5: 0.0, joint6: 1.0}"

#Return to home position:
rosrun kinova_demo joints_action_client.py -v j2n6s300 degree -- 275 167 57 -119 82 435

#Moves fine in cartesian:
rostopic pub -r 100 /j2n6s300_driver/in/cartesian_velocity kinova_msgs/PoseVelocity "{twist_linear_x: 1.0, twist_linear_y: 0.0, twist_linear_z: 0.0, twist_angular_x: 0.0, twist_angular_y: 0.0, twist_angular_z: 0.0}" 
`

It's clear that this is position dependent.  Is there anyway I can bypass the "swap to angular control when near a singularity/collision/etc."?

alexvannobel, 2019-10-30:
Hi @mfry90 ,

Sorry for the delay. I have had some time to reproduce your issue, and found out as you did that calling the joints_action_client action server sometimes fails to put back the arm in Cartesian mode, which makes the cartesian_velocity commands silently fail afterwards.
I've tried to automatically call setCartesianControl when the action ends, but the mode didn't change (as you saw). The error code from the API didn't indicate anything fishy, so I opted for a getCartesianPosition + setCartesianPosition to the current position. 
It correctly switches the mode to Cartesian, because the joystick commands I send move the arm in Cartesian space now. 
However, I still have a problem sending Cartesian velocities after I did that. The Cartesian velocities I send right after the end of the angular action don't move the arm, but as soon as I send a joystick command, the cartesian velocities start working again. 
I'll push what I have right now in a branch and dig in the arm's firmware to find an API solution to this problem.

I'll keep you updated,
Alex

alexvannobel, 2019-10-30:
Hi again @mfry90 ,

By what I see in the arm's firmware, if the arm is in a singularity, it won't be able to switch to Cartesian velocity control mode and will then fallback to angular velocity control mode. I can see this behaviour if I run 
```
rosrun kinova_demo joints_action_client.py -v j2n6s300 degree -- 270 225 109 -4.3 32 437
rostopic pub -r 100 /j2n6s300_driver/in/cartesian_velocity kinova_msgs/PoseVelocity "{twist_linear_x: 0.0, twist_linear_y: 0.0, twist_linear_z: 1.0, twist_angular_x: 0.0, twist_angular_y: 0.0, twist_angular_z: 0.0}"
```
The arm doesn't move with the second command. This also explains why the joystick commands move the arm in joint space. It needs to be homed back before moving in Cartesian space.
However, if I run: 
```
rosrun kinova_demo joints_action_client.py -v j2n6s300 degree -- 270 270 109 -4.3 32 437
rostopic pub -r 100 /j2n6s300_driver/in/cartesian_velocity kinova_msgs/PoseVelocity "{twist_linear_x: 0.0, twist_linear_y: 0.0, twist_linear_z: 1.0, twist_angular_x: 0.0, twist_angular_y: 0.0, twist_angular_z: 0.0}"
```
the arm is not in a singularity and is able to switch to Cartesian velocity control. It moves slowly at first, but it reaches the desired speed as it finds the solution.
I'll ask colleagues tomorrow if there is another way to switch to Cartesian mode, but I'm beginning to think the arm is simply unable to compute the Cartesian solution to the velocity commands, being in a singularity.

I'll keep you updated tomorrow,
Best,
Alex

alexvannobel, 2019-10-31:
Hi @mfry90 ,

I was confirmed the behaviour seen here is normal for the Jaco2 arm; the Cartesian solver in the robot's base cannot converge to a solution, so the arm won't move with Cartesian Velocity commands. 
This is also why the arm won't switch to Cartesian control mode: if it's in a singularity, or too close to one, it will fallback on angular control mode.
When you get in these positions, you could try to move slightly in joint space until you leave the singularity so you can send Cartesian commands to the arm. You could also use MoveIt to plan a Cartesian command (sent eventually to the arm as joint speeds) to get you out of the singularity.

Let me know what you think,
Alex

mfry90, 2019-10-31:
Thanks @alexvannobel for the response.  Everything you've stated is as I observe as well.  What I don't understand is how the arm responds to cartesian velocity commands sent via the _joystick_, but not ROS.  As we've both observed, we can switch it to cartesian mode, and the joystick will move it but ROS cartesian velocity commands won't.  Typically though, AFTER issuing a brief joystick command in cartesian mode, the arm responds to ROS cartesian velocity commands.  

Is there some way I can mimic the joystick via ROS? Clearly the arm is able to move in Cartesian around a singularity via the joystick, so if I could send a "joystick-looking" command via ROS, I imagine I could get the system to respond to typical ROS cartesian velocity commands again.

alexvannobel, 2019-10-31:
Hi @mfry90 ,

The embedded joystick control is more permissive than the API Cartesian velocity commands, and make it easier to leave the singularity states. This is a particularity that was needed (the joystick being more permissive) for Assistive users, that really can't get stuck with the arm, and who only have the joystick to control it. Also, when using the joystick, we kind of  "feel" what's the best direction to take and command that way. We can't do that with the API, we can only send a fixed speed. 

However, if you want to mimic a joystick command, you can do it with the API with the SendJoystickCommand function. This is a function in the [kinova_api class](https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_driver/src/kinova_api.cpp#L79), but it's not exposed right now in the kinova_comm object. You could create a Topic in the kinova_arm_driver that could call the kinova_comm object to send a joystick command to the arm, if you really wanted to try the joystick commands. If you want to go up/down in Z, I believe it's the [Rotate member](https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_driver/include/kinova/KinovaTypes.h#L910) of the JoystickCommand struct you would have to fill with the float you want.

Hope that helps,
Alex

mfry90, 2019-11-05:
Got it.  I ended up modifying kinova_arm.cpp and kinova_comm.cpp to subscribe to a "virtual joystick topic" and send the virtual joystick commands via the kinova_api class.

alexvannobel, 2019-11-05:
Hi @mfry90 ,

Glad you were able to send joystick commands! I'll close this issue then, but if you wish to pull request your change to the driver I'll be glad to review it.

Best,
Alex

closed 2019-11-05 by alexvannobel