Kinovarobotics/kinova-ros #270: configuration about torque sensor in gazebo

ZiQiangZhou, 2019-09-27:
version: ubuntu 16.04
ros: kinematic
gazebo: 7.0.0

i want to get the torque sensor data when i do some simulation in gazebo. but i hasn't found any topics about that after launching the "kinova_gazebo/launch/robot_launch.launch ".

refer to [this](https://answers.ros.org/question/221702/force-torque-sensor-simulation/),  i added some the code like below in j2s7s300.xacro, but there was always 0 for the ft_sensor_topic topic 
`
      
    <gazebo>
        <plugin name="ft_sensor" filename="libgazebo_ros_ft_sensor.so">
            <updateRate>100.0</updateRate>
            <topicName>ft_sensor_topic</topicName>
            <jointName>j2s7s300_joint_1</jointName>
            <noise>
                <type>gaussian</type>
                <mean>0.0</mean>
                <stddev>0.003</stddev>  
            </noise>            
        </plugin>
    </gazebo>
`
i have also tried like [this](http://gazebosim.org/tutorials?tut=force_torque_sensor&cat=sensors),  add the code below in kinova_common.xacro, but there was also no topic when i launched "kinova_gazebo/launch/robot_launch.launch", and there was no data when i run `gz topic -e /gazebo/default/j2s7s300/j2s7s300_link_1/wrench ` .
i am the beginner of ros and gazebo, i have tried my best but i still don't know how to get the torque sensor data, could u help me? thanks!

alexvannobel, 2019-10-31:
Hi @ZiQiangZhou ,

Sorry for the delay. I have had time now to look at your issue, and found a solution on [this post](https://answers.ros.org/question/221702/force-torque-sensor-simulation/). I added the plugin to the arm joints in the xacro and made a PR.
Essentially, with the change, you can access the force/torque information in the "/{your_robot_type}_joint_{joint_index}_ft_sensor_topic". So, to access the data of actuator 1 for j2s7s300, you would have to subscribe to the "/j2s7s300_joint_1_ft_sensor_topic".

I'll let you know when this is merged,
Best regards,
Alex

alexvannobel, 2019-11-01:
Hi @ZiQiangZhou ,

The PR is merged. I'll close the issue, but don't hesitate to reopen it if you have problems getting the wrench for the joints in simulation.

Best,
Alex

closed 2019-11-01 by alexvannobel