Kinovarobotics/kinova-ros #271: Cmake error in Catkin_make for melodic

rachelhson, 2019-10-09:
## Description

Catkin_make is failed because of the issue see ## Code example.
I am not sure how I could fix this problem. I guess inverse_kinematics_plugins could not find the package from python27amd64. How could I add packages in cmake ? 

## Version

ROS distribution : melodic 

Branch and commit you are using :  master 

## Steps to reproduce

1. *step 1...*
2. *step 2...*

## Code example (if necessary)

```
CMake Error at C:/opt/python27amd64/Lib/site-packages/cmake/data/share/cmake-3.15/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find BLAS (missing: BLAS_LIBRARIES)
Call Stack (most recent call first):
  C:/opt/python27amd64/Lib/site-packages/cmake/data/share/cmake-3.15/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  C:/opt/python27amd64/Lib/site-packages/cmake/data/share/cmake-3.15/Modules/FindBLAS.cmake:806 (find_package_handle_standard_args)
  C:/opt/python27amd64/Lib/site-packages/cmake/data/share/cmake-3.15/Modules/FindLAPACK.cmake:193 (find_package)
  kinova-ros/kinova_moveit/inverse_kinematics_plugins/ikfast/j2n6s300_ikfast/CMakeLists.txt:28 
(find_package)

-- Configuring incomplete, errors occurred!
See also "C:/opt/ros/melodic/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "C:/opt/ros/melodic/catkin_ws/build/CMakeFiles/CMakeError.log".
NMAKE : fatal error U1077: 'C:\opt\python27amd64\Lib\site-packages\cmake\data\bin\cmake.exe' : return code '0x1'
Stop.
Invoking "nmake cmake_check_build_system" failed

```

## Expected behavior

successfully done catkin_make

## Any other information

I installed ros-melodic in window 10.

rachelhson, 2019-10-11:
The issue is resolved by `choco install openblas -y`. 

now I have another issue 
```
 85%] Building CXX object kinova-ros/kinova_driver/CMakeFiles/kinova_driver.dir/src/kinova_api.cpp.obj
kinova_api.cpp
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(313): error C3861: 'visibility': identifier not found
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(313): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(313): error C2448: '__attribute__': function-style initializer appears to be a function definition
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(315): error C3861: 'visibility': identifier not found
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(315): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(315): error C2374: '__attribute__': redefinition; multiple initializationC:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(313): note: see declaration of '__attribute__'
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(315): error C2448: '__attribute__': function-style initializer appears to be a function definition
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(317): error C3861: 'visibility': identifier not found
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(317): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(317): error C2374: '__attribute__': redefinition; multiple initializationC:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(313): note: see declaration of '__attribute__'
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(317): error C2448: '__attribute__': function-style initializer appears to be a function definition
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(319): error C3861: 'visibility': identifier not found
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(319): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(319): error C2374: '__attribute__': redefinition; multiple initializationC:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(313): note: see declaration of '__attribute__'
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(319): error C2448: '__attribute__': function-style initializer appears to be a function definition
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(323): error C3861: 'visibility': identifier not found
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(323): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(323): error C2374: '__attribute__': redefinition; multiple initializationC:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(313): note: see declaration of '__attribute__'
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(323): error C2146: syntax error: missing ';' before identifier 'Packet'
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(325): error C3861: 'visibility': identifier not found
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(325): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(325): error C2374: '__attribute__': redefinition; multiple initializationC:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(313): note: see declaration of '__attribute__'
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(325): error C2448: '__attribute__': function-style initializer appears to be a function definition
C:\opt\ros\melodic\catkin_ws\src\kinova-ros\kinova_driver\include\kinova\Kinova.API.USBCommLayerUbuntu.h(327): error C3861: 'visibility': identifier not found
[... cut ...] tx64\x64\cl.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.23.28105\bin\HostX64\x64\nmake.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.23.28105\bin\HostX64\x64\nmake.exe"' : return code '0x2'
Stop.
Invoking "nmake" failed
```

I was trying to find `USBCommandLayer.h` for window, but i cannot find one in github.

alexvannobel, 2019-10-31:
Hi @rachelhson ,
Sorry for the delay. We don't support ROS for Windows, as the Kinova API included with the package is built for a Unix architecture. We don't plan on adding support for ROS Windows in kinova-ros, either.
If you really want to be on a Windows machine, I guess you could try to run inside a Docker container  or a VM, but running in the Windows OS surely won't work.

Best,
Alex

closed 2019-10-31 by alexvannobel