Kinovarobotics/kinova-ros #272: Joint Velocity and Torque limits for the 7DoF robot

EceChaik, 2019-10-11:
Guys, you have not included "joint_velocity_limit" and "joint_torque_limit" terms in the j2n7s300.xacro file, for the 7DoF robot. 
Is it a good assumption to use the same values from the j2n6s300 ?

alexvannobel, 2019-10-31:
Hi @EceChaik ,

Really sorry for the delay. You are right, we didn't specify it for the j2n7s300, we missed it. The non-spherical wrists 7DOFs robots are very rare though, we didn't make much. Is that what you have?
Normally, we make sperical wrists 7DOF robots (j2s7s300).
The limits for the j2s7s300 are the same as the limits for the j2n7s300. 
Anyways, I fixed that in #283 . I'll let you know when it's merged. 

Best,
Alex

EceChaik, 2019-10-31:
Hi Alex, after opening the issue I noticed that indeed what I wanted was the spherical wrist robot, but nevertheless thank you.

closed 2019-11-01 by alexvannobel