Kinovarobotics/kinova-ros #274: USBCommandLayerUbuntu.so Issue

jeeseop, 2019-10-15:
## Description
I am using TX2 with j2n4s300 ver. Kinova arm.
I want to control it through the usb and ethernet cable, but both didn't work.
I cross-checked with normal ubuntu laptop, and I figured out that every comm(command)layerubuntu.so(USB and ETH)  file is not correctly generated in the /usr/lib directory.

Is there any detour that I can install this libraries in the correct position?
## Version

ROS distribution : 16.04 Kinetic

Branch and commit you are using : master

## Steps to reproduce

1. Using TX2

LazyEngineerToBe, 2019-10-22:
You could try the following: https://github.com/Kinovarobotics/Extra_code_examples_Jaco2arm/tree/master/sdk%20compiled%20for%20tx2

alexvannobel, 2019-10-31:
Hi @jeeseop ,

Sorry for the delay. The library files that come with the ROS package are compiled for x86-64, not for ARM. However, as @LazyEngineerToBe said, you could try to use the library compiled for TX2 (but it is not officially supported by Kinova).
Did you try it and did you make it work?

Best,
Alex

alexvannobel, 2020-01-28:
Hi @jeeseop ,

I'll close this issue for inactivity, but feel free to re-open it if it persists or to open a new issue.

Best regards,
Alex

closed 2020-01-28 by alexvannobel