Kinovarobotics/kinova-ros #276: Can not get to the target in moveit with real Jaco2

flymaxty, 2019-10-21:
## Description

Hi all, sorry for my poor english.
I'm trying to use Kinova Jaco v2 to do pick & place, I'm using moveit to control real kinova arm.
But I found that the arm always can not get to / overshoot to the target pose. Looks like trajectory-controller stop the arm on incorrect time. 

## Version

Arm: Kinova Jaco2, j2n6s300
ROS distribution : Melodic
Branch and commit : master, latest commit

mpatalberta, 2019-10-26:
You are possibly running into the callback timer not running on the required 0.010 interval.
The time slippage results in missapplication of velocities

alexvannobel, 2019-11-01:
Hi @flymaxty ,

@mpatalberta has fixed the timer callback timing, so when I'll have the code reviewed, tested and merged, I'll be able to test the MoveIt targets and see if the timer solves the situation (although it seems like it doesn't, as of #279 ).
I'll investigate when I have those changes merged, so I'll update you then.

Thanks for your patience, 
Best,
Alex

flymaxty, 2019-11-02:
Hi @mpatalberta and @alexvannobel,
Thanks for your reply!
Looking forward to your updates :）

Best,
Ye

flymaxty, 2020-06-10:
Hi @alexvannobel ,
any updates about this issue? I also found that this repo has brach named "improved_precision_moveit_plan" and "reduce-overshoot-follow-joint-trajectory".

Are those branches releated with this issue?

closed 2021-06-04 by felixmaisonneuve