Kinovarobotics/kinova-ros #278: Can Finger position control and Velocity Control for Cartesian space run at the same time

lienbo, 2019-10-25:
## Description
I publish cartesian_velocity and finger position control like fingers_action_client.py at the same time, but can't run at the same time. My method is to add time.sleep() in one control loop. It's not smooth.
One paragraph to explain the issue

## Version

ROS distribution : 

Branch and commit you are using : 

## Steps to reproduce

1. *step 1...*
2. *step 2...*

## Code example (if necessary)

```
    # paste code here #
```

## Expected behavior

Describe the expected behavior (in regard to the bug described above).

## Any other information

Any other information you believe the maintainers need to know.

alexvannobel, 2019-10-31:
Hi @lienbo ,

It's actually impossible the way the driver is currently designed to send finger commands and cartesian velocity commands at the same time, because the driver has a queue for trajectories and no callback contains information for both cartesian velocities and finger positions. I know it's possible in the API, though, so I tested by adding a msg that contains both the cartesian velocity and the finger positions and managed to create the necessary topic and function in the driver to send it all at the same time. 
This way, I can manage to move the fingers and the arm at the same time, but I saw that if the arm command was a velocity command, the fingers's command is also interpreted as a velocity command, so what you want to do is possible only if you send a cartesian position to the arm at the same time as a finger position.
Bottom line: you cannot send finger **positions** and arm **velocities**. You can only send finger **positions** and arm **positions**, or finger **velocities** and arm **velocities**.
What's your thought on that? Would you like me to post my code so you can see how I did?

Best,
Alex

lienbo, 2019-11-01:
> Hi @lienbo ,
>
> It's actually impossible the way the driver is currently designed to send finger commands and cartesian velocity commands at the same time, because the driver has a queue for [...]

Thanks for your reply

So the best way for me is to give up the kinova-ros, using the low level api instead .

alexvannobel, 2019-11-01:
Hi  @lienbo ,

No need to give up on the ROS repo!
I added some code to support sending finger position + cartesian velocities at the same time finally, I just made a PR.
Can you try it? I made a small proportional controller to make sure the fingers, even if they are controlled in velocity, reach the target position, in closure percentage (0 = opened, 100 = closed).
I'll wait for your feedback, and let you know when it's merged. 

Best,
Alex

alexvannobel, 2019-11-27:
Hi @lienbo , 
I just merged the pull request to `master`. 
I'll close this issue for now, but don't hesitate to reopen the issue if you encounter the problem again on the `master` branch. 

Best,
Alex

closed 2019-11-27 by alexvannobel