Kinovarobotics/kinova-ros #279: Timer callback does appear at regular intervals

mpatalberta, 2019-10-26:
## Description
The timer callback is not called at a regular 0.01 second intervals.
This results in misapplication of the location when using a velocity trajectory baes
One paragraph to explain the issue

## Version
current tip
ROS distribution : 
melodoc
Branch and commit you are using : 
tip
## Steps to reproduce

1. Create an array the holds the times
2. Run a locatiom
3 at the end printout all the times in the array

## Code example (if necessary)

```
    # paste code here #
```

## Expected behavior

Describe the expected behavior (in regard to the bug described above).

## Any other information

Any other information you believe the maintainers need to know.

mpatalberta, 2019-10-26:
You could try to replace the ros timer callback with linux pthread timer callback.

mpatalberta, 2019-11-01:
I have changed the software so the software generates timer at 0.010 seconds without a lot of slip.
I also change the code to directly call the kinova instead of going through a publish and advertise.

This did however solve the problem that the moveit do not do a complete path.
You still need to call moveit more than once.

mpatalberta, 2019-11-01:
do you want the code change?
where do I send it.

mpatalberta, 2019-11-01:
woops my fault

alexvannobel, 2019-11-01:
Hi @mpatalberta ,

Sorry for the delay.
Good news! Thanks for the contribution. I'm guessing you forked kinova-ros and created a branch to write that fix. You can simply go to [Pull Requests](https://github.com/Kinovarobotics/kinova-ros/pulls) and create a new pull request from your repo's forked branch to our master branch. 
I'll have a look at it when it's created.
So you're saying now the thread correctly runs at 100Hz, but the MoveIt trajectories still do not exactly reach the target point?

Best,
Alex

mpatalberta, 2019-11-01:
Alex,
I need to clean up the code a bit.
The attached spreadsheet have the results.
Rth is the old timer callback
Pth is the pthread timer callback.
The times are recorded in a local buffer and then dumped at the end.
All printout are removed.
The times are correct but still the moveit does not get completely to the final location.
Pat,

[External Email]

- CONFIDENTIAL-

This email and any files transmitted with it are confidential, and may also be legally privileged. If you are not the intended recipient, you may not review, use, copy, or distribute this message. If you receive this email in error, please notify the sender immediately by reply email and then delete this email.

alexvannobel, 2019-11-01:
Hi @mpatalberta ,

There doesn't seem to be an attached file. It's ok, you can take your time and I'll test and review the code when you post it. I'll investigate the MoveIt also when the code is available. 

Thanks again,
Best,
Alex

mpatalberta, 2019-11-01:
Alex,
Do you have a real email address I will send the excel spreadsheet.
Also I am thinking that inorder to get the to the final position we should angular velocity to simply angle.
Pat,

mpatalberta, 2019-11-03:
OLD TIMER
avg | 0.01005477 |  
stddev*3 | 0.00699531 | 0.02098592

New timer
avg | 0.00999996 |  
stddev*3 | 0.00018873 | 0.00056619

The new timer  has a much better sigma. 
Which implies a better timing interval

alexvannobel, 2019-11-04:
Hi @mpatalberta ,

Yes, it seems much better! You can send the spreadsheet at ros@kinova.ca.
Or, to be sure to reach the final position, we could control in angular velocity to reach the mid-points and in angular position to reach the endpoint (I would have to test that the arm doesn't stop though when changing the mode).

Best,
Alex

mpatalberta, 2019-11-04:
Rth – ros timer callback
                So you know it is mentioned that the ros timer callback is not reliable
Pth – pthread timer callback

Patrick Brown
631 880 1188

Yes, it seems much better! You can send the spreadsheet at ros@kinova.ca<mailto:ros@kinova.ca>.

closed 2021-06-02 by felixmaisonneuve