Kinovarobotics/kinova-ros #281: How to change the topic publish rate?

lienbo, 2019-10-27:
## Summary
I use the commend` rostopic hz /.../.../joint_state`  and `/.../.../tool_pose` and I know the default publish rate is only 10hz, but the Cartesian space velocity control needs a rate of 100hz which needs the msgs of tool_pose and joint_state. So how to do that?

alexvannobel, 2019-11-01:
Hi @lienbo ,

The `joint_state` and `tool_pose` topics are published as part of the [status_timer](https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_driver/src/kinova_arm.cpp#L690-L696) function, which is triggered by the [status_timer_](https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_driver/src/kinova_arm.cpp#L228-L229) timer, driven by the `status_interval_seconds` ROS param, defaulting to 0.1.
I submitted a PR to add this param to the launch file, so you can simply specify it in the launch file like so : 
`roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=j2n6s300 feedback_publish_rate:=0.01`

Best,
Alex

alexvannobel, 2019-11-01:
However, I just tested the `rostopic hz /j2n6s300_driver/out/tool_pose` command when I specify the interval as 0.01 second, and I can barely reach 40Hz (and it is supposed to be 100).
This is because every call to the Jaco C++ API takes some time to return, and [the four publish...() functions called by the timer](https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_driver/src/kinova_arm.cpp#L692-L695) each make calls to this API. The [publishJointAngles function](https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_driver/src/kinova_arm.cpp#L499-L631) is the one with the most calls. The feedback was not designed to run at 100Hz, so if you want to reach 100Hz on ROS feedback, you should consider commenting out the calls to the kinova_comm_ object that are not necessary for your application in the publish...() functions.

Hope this helps,
Alex

alexvannobel, 2019-11-01:
Hi @lienbo ,
The PR is merged so you can test it on the `master` branch.
Best,
Alex

alexvannobel, 2020-01-28:
Hi @lienbo  ,

Since the PR is merged and I haven't heard from you, I'll close this issue for inactivity but feel free to re-open it if it persists or to open a new issue.

Best regards,
Alex

closed 2020-01-28 by alexvannobel