Kinovarobotics/kinova-ros #286: Moveit for Kinova in Ubuntu 18.04

kurtserP, 2019-11-06:
## Description
Running the demo launch files for moveit! under ROS melodic (Ubuntu Bionic 18.04) using the following line:
`  roslaunch j2n6s300_moveit_config j2n6s300_virtual_robot_demo.launch`
Results in the following error:
```
[ERROR] [1573036239.753641206]: Unable to parse component [.649262481663582] to a double (while parsing a vector value)
[ERROR] [1573036239.753710501]: Malformed parent origin element for joint [j2n6s300_joint_finger_1]
[ERROR] [1573036239.753750440]: joint xml is not initialized correctly
[ERROR] [1573036239.754074436]: Unable to parse URDF from parameter '/robot_description'
[ERROR] [1573036239.758041714]: Robot model not loaded
```

similar error appears for  j2s7s300 and  j2s6s300. For j2s6s300 rviz loads the end effector frame, while for the rest an empty rviz workspace appears.

Attempted to run the same for a read robot j2n6s300 using:
`roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=j2n6s300` 
and
`roslaunch j2n6s300_moveit_config j2n6s300_virtual_robot_demo.launch`
results in the same issue, though communication with the robot was indeed established (first line runs fine and the robot stiffens the gripper).

Any ideas? Followed the basic installation documentation.

alexvannobel, 2019-11-13:
Hi @kurtserP ,

I just tested with ROS Kinetic and I have no trouble launching 
`roslaunch j2n6s300_moveit_config j2n6s300_virtual_robot_demo.launch`
I can move the arm in RViz without trouble.
I'll get my hands on a computer with ROS Melodic and I'll test later this afternnoon.

Best,
Alex

alexvannobel, 2019-11-13:
Hi @kurtserP ,
I tested on ROS Melodic and I had no problem launching
`roslaunch j2n6s300_moveit_config j2n6s300_virtual_robot_demo.launch`
Could that be a namespace error if `robot_description` has been remapped? Did you source the workspace correctly before launching?

Let me know,
Best,
Alex

alexvannobel, 2020-01-28:
Hi @kurtserP  ,

I'll close this issue for inactivity, but feel free to re-open it if it persists or to open a new issue.

Best regards,
Alex

closed 2020-01-28 by alexvannobel