Kinovarobotics/kinova-ros #287: Joint admittance control not working

LongfeiProjects, 2019-11-20:
I am playing around with force/torque control with Jaco2, and I found joint admittance control not working, while Cartesian admittance control works fine.

I may also take the chance to clarify some concepts regarding to force/torque control in Jaco2:
1. Jaco2 has two "torque" control mode:  joint torque control and Cartesian force control
Both torque control mode can be activated by rosservice: in/set_torque_control_mode 1. In joint or Cartesian torque control mode depends on which topic you are commanding: in/joint_torque or in/cartesian_force

2. Jaco2 has two "force" control mode: joint admittance control and Cartesian admittance control
Both admittance control can be activated by rosservice: in/start_force_control. In joint or Cartesian admittance mode depends on the moment before "start_force_control". 

The following point describes the problem I met:
3. When I am in Cartesian mode, I can enter Cartesian admittance mode by "start_force_control".  If I am in Joint mode and call "start_force_control", it failed to enter joint admittance mode(simply nothing happened). Then I switch to Cartesian mode, Cartesian admittance still works. The key commands are:
$ roslaunch kinova_bringup kinova_robot.launch
$ rosrun kinova_demo joints_action_client.py j2n6s300 degree -- 180 180 180 180 180 180
$ rosservice call /j2n6s300_driver/in/set_zero_torques
$ rosservice call /j2n6s300_driver/in/start_force_control

I tried other positions, setting parameters as well, but still joint admittance not working. I suppose it is not ROS related, since using DevelopmentCenter SDK and given "Admittance" example code shows the same situation. Any clue on this issue?

Serial No.: PJ00900006510709-01
Main firmware: 6.2.5.0
Actuators: 2.1.1
Fingers: 1.1.4
Comm interface1: 1.0.11
Comm interface2: 0.0.0

LongfeiProjects, 2019-11-20:
I confirmed on the other arm that it works.

closed 2019-11-20 by LongfeiProjects

TheHidj, 2022-02-09:
Hello, 
What values did you use for the Cartesian admittance control ? Did you use the set_force_params service ? 
Thank you ,