Kinovarobotics/kinova-ros #288: IKFast plugin for the j2s7s300 arm

mabouseif, 2019-11-21:
Hello,

I am trying to do (Deep) Reinforcement Learning with the j2s7s300, and i need to do tens of thousands of iterations, and currently both the KDL and TRAC IK require a very long time for computing a solution, which is not suitable for my application.
On average, they both take multiple orders of magnitude longer, but they give the correct solution, while the IKFast plugin fails to do so.

Is the __correct__ IKFast plugin ever going to be implemented?

According to you, it is giving only the elbow down solution. Any idea what can be done about that?

Thanks : )

alexvannobel, 2019-11-25:
Hi @mabouseif ,

Unfortunately, no, we did not work on IKFast since. We did not acquire more knowledge on IKFast either internally. I can create a request to dig the subject, regenerate and test IKFast but it most likely won't be done soon.  
Are you able to regenerate the IKFast plugin yourself with the most recent URDF for the Jaco2 7DOF?

Best,
Alex

mabouseif, 2019-11-26:
Hi @alexvannobel 

I have not tried to generate it yet. I will try to do so at some point. If you manage to do it from your side that would be great, since i do not have experience with IKFast. I will update here if i reach anything.

Thanks : )

closed 2021-06-02 by felixmaisonneuve