Kinovarobotics/kinova-ros #290: Can't roslaunch bringup Kinova JACO arms anymore

blocka23, 2019-11-27:
## Description

Hello, I recently upgraded to ROS Kinetic on Ubuntu 16.04. Now, unfortunately, I'm getting an error when I try to roslaunch bringup my Kinova JACO arms. I've tried googling for the answer with no luck of anyone else facing this issue. Thanks very much in advance for any advice or suggestions you provide to helping me get my arms back up and running again!

## Version

ROS distribution : Kinetic

Branch and commit you are using : master

## Code example (if necessary)

```
    roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=j2n6s300 kinova_robotName:=left kinova_robotSerial:=PJ00900006WO5044820

```

## Expected behavior

In this past, this code working for bringing up the arms and there'd be no issues and I could just go about with controlling htem.

## Now I'm getting these errors --> see the bottom especially [FATAL] and [left_driver-1]

```
PARAMETERS
 * /kinova_number_of_robots: 2
 * /kinova_robots: [{'serial': 'PJ00...
 * /left_driver/connection_type: USB
 * /left_driver/ethernet/local_broadcast_port: 25025
 * /left_driver/ethernet/local_cmd_port: 25000
 * /left_driver/ethernet/local_machine_IP: 192.168.100.100
 * /left_driver/ethernet/subnet_mask: 255.255.255.0
 * /left_driver/jointSpeedLimitParameter1: 10
 * /left_driver/jointSpeedLimitParameter2: 20
 * /left_driver/robot_name: left
 * /left_driver/robot_type: j2n6s300
 * /left_driver/serial_number: PJ00900006WO5044820
 * /left_driver/torque_parameters/publish_torque_with_gravity_compensation: False
 * /left_driver/torque_parameters/use_estimated_COM_parameters: False
 * /left_driver/use_jaco_v1_fingers: False
 * /robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    left_driver (kinova_driver/kinova_arm_driver)
    left_state_publisher (robot_state_publisher/robot_state_publisher)

ROS_MASTER_URI=http://localhost:11311

process[left_driver-1]: started with pid [3364]
process[left_state_publisher-2]: started with pid [3365]
[ INFO] [1574859834.374096419]: kinova_robotType is j2n6s300.
[ INFO] [1574859834.375342370]: kinova_robotName is left.
[FATAL] [1574859834.378008786]: USBCommandLayerUbuntu.so: cannot open shared object file: No such file or directory
kinova_arm_driver: /home/alexis/catkin_ws/src/kinova-ros/kinova_driver/src/kinova_api.cpp:26: void* kinova::KinovaAPI::initCommandLayerFunction(const char*): Assertion `function_pointer != NULL' failed.
[left_driver-1] process has died [pid 3364, exit code -6, cmd /home/alexis/catkin_ws/devel/lib/kinova_driver/kinova_arm_driver j2n6s300 __name:=left_driver __log:=/home/alexis/.ros/log/467f01d0-1116-11ea-8784-8ca982d60e8e/left_driver-1.log].
log file: /home/alexis/.ros/log/467f01d0-1116-11ea-8784-8ca982d60e8e/left_driver-1*.log

```

brenhertel, 2022-09-07:
What was the resolution to this error? I am running into the same issue.

brenhertel, 2022-09-13:
Hi Felix,

This is the case.

After setting up the repo using the instructions listed at https://github.com/Kinovarobotics/kinova-ros I first tried the kinova_robot.launch command and got this error. I've tried several fixes, such as specifically including these folders in $LD_LIBRARY_PATH, using ldconfig, and moving the files into /lib, all of which ran into the same errors. Additionally I've cleaned and used make/build on the workspace multiple times.

Thanks for your help,
Brendan

felixmaisonneuve, 2022-09-14:
Was it working before? If so, what did you change?

Are you running ROS on a raspberry pi or other ARM computer?

brenhertel, 2022-09-21:
I have this working on other computers and have never run into this issue before. This issue arose on a computer running ARM, specifically the NVIDIA Jetson Tx2.

felixmaisonneuve, 2022-09-22:
@brenhertel, this is probably the issue. 
I can refer you to the [build section](https://github.com/Kinovarobotics/ros_kortex#build) of the main readme, where yu can specify wich architecture you are using. 

Since you are using Jetson Tx2, you should build you ROS stack with `catkin_make --cmake-args -DCONAN_TARGET_PLATFORM=jetson`

Try it out and tell me if it works
Best,
Felix

brenhertel, 2022-09-22:
Hi Felix,

I was able to successfully install and build according to the instructions at that link, but unfortunately the error persists. I will post the output I am getting for reference in case it helps.

```
$ roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=j2s7s300
... logging to /home/ghost/.ros/log/1d3b392c-39ce-11ed-b2da-000c8bb4112b/roslaunch-ghost-15943.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro.py is deprecated; please use xacro instead
started roslaunch server http://ghost:35823/

SUMMARY
========

PARAMETERS
 * /j2s7s300_driver/connection_type: USB
 * /j2s7s300_driver/ethernet/local_broadcast_port: 25025
 * /j2s7s300_driver/ethernet/local_cmd_port: 25000
 * /j2s7s300_driver/ethernet/local_machine_IP: 192.168.100.100
 * /j2s7s300_driver/ethernet/subnet_mask: 255.255.255.0
 * /j2s7s300_driver/jointSpeedLimitParameter1: 10
 * /j2s7s300_driver/jointSpeedLimitParameter2: 20
 * /j2s7s300_driver/robot_name: j2s7s300
 * /j2s7s300_driver/robot_type: j2s7s300
 * /j2s7s300_driver/serial_number: not_set
 * /j2s7s300_driver/status_interval_seconds: 0.1
 * /j2s7s300_driver/torque_parameters/publish_torque_with_gravity_compensation: False
 * /j2s7s300_driver/torque_parameters/use_estimated_COM_parameters: False
 * /j2s7s300_driver/use_jaco_v1_fingers: False
 * /kinova_number_of_robots: 2
 * /kinova_robots: [{'serial': 'PJ00...
 * /robot_description: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.13

NODES
  /
    j2s7s300_driver (kinova_driver/kinova_arm_driver)
    j2s7s300_state_publisher (robot_state_publisher/robot_state_publisher)

ROS_MASTER_URI=http://localhost:11311

process[j2s7s300_driver-1]: started with pid [15975]
process[j2s7s300_state_publisher-2]: started with pid [15976]
[ INFO] [1663869660.624143137]: kinova_robotType is j2s7s300.
[ INFO] [1663869660.638073953]: kinova_robotName is j2s7s300.
[FATAL] [1663869660.643653396]: USBCommandLayerUbuntu.so: cannot open shared object file: No such file or directory
kinova_arm_driver: /home/ghost/catkin_ws/src/kinova-ros/kinova_driver/src/kinova_api.cpp:26: void* kinova::KinovaAPI::initCommandLayerFunction(const char*): Assertion `function_pointer != NULL' failed.
[j2s7s300_driver-1] process has died [pid 15975, exit code -6, cmd /home/ghost/catkin_ws/devel/lib/kinova_driver/kinova_arm_driver j2s7s300 __name:=j2s7s300_driver __log:=/home/ghost/.ros/log/1d3b392c-39ce-11ed-b2da-000c8bb4112b/j2s7s300_driver-1.log].
log file: /home/ghost/.ros/log/1d3b392c-39ce-11ed-b2da-000c8bb4112b/j2s7s300_driver-1*.log
^C[j2s7s300_state_publisher-2] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

felixmaisonneuve, 2022-09-28:
Hi @brenhertel,

I am very sorry, my previous comment is wrong. I sent you to our Gen3 page, but you are using a Jaco2. 
However, the issue remains the same. Your library files (.so files) are compiled for the x86 architecture, but you are using an ARM architecture.

We do not provide official library files for other system architecture.
BUT, some work was done in the past to compile the API on different architectures : 
- https://github.com/Kinovarobotics/Extra_code_examples_Jaco2arm/tree/master/sdk%20compiled%20for%20tx2 
- https://github.com/Kinovarobotics/kinova_sdk_recompiled/tree/master/raspberry%20pi%203

This is given "as is" and I have no garanty if it will work or not. 

Best of luck,
Felix

brenhertel, 2022-10-06:
Hi Felix,

Thank you, I think this has been a huge step in the right direction. Using the first link you posted (the one compiled for Jetson Tx2), I downloaded the .so files and renamed them to the names of the libraries of the original kinova driver (https://github.com/Kinovarobotics/kinova-ros/tree/melodic-devel/kinova_driver/lib). It seems to actually be able to open these libraries now and attempt to run them. Unfortunately, a new error has arisen:
`[FATAL] [1664478252.359851809]: /home/ghost/catkin_ws/src/kinova_libs/USBCommLayerUbuntu.so: undefined symbol: libusb_close`
libusb-1.0 is installed, so I am unsure why this symbol would appear undefined. If you have any suggestions it would be appreciated, although I fear that this may be outside the scope of the original error in the thread.

Thanks,
Brendan

felixmaisonneuve, 2022-10-11:
This seems to be an error with the libusb package.
Maybe update / upgrade your system `sudo apt update && sudo apt upgrade`

If it doesn't work I would try to remove and re-install different packages : libusb, libusb-1.0, libusb-dev or libusb-1.0-0-dev

felixmaisonneuve, 2022-12-15:
Hi @blocka23,

Did you manage to resolve the issue?
Can we close this?

I will close the issue regardless in a couple of days/weeks if I do not get any update.

Best,
Felix