Kinovarobotics/kinova-ros #291: can Gazebo simulation use kinova_demo action client ?

rachelhson, 2019-12-13:
## Summary
The goal of feature is simulating the arm motion same as the real motion. 

## Use case
- Why is this feature needed?
Virtual environment testing is quick and easier to finish without connecting to a real robot. 
Currently, I believe the Jaco arm is not able to test with the same code that the real robot use. 

- What are the use case covered by this feature?
It can also test with some sensor on it, we can simulate the expected arm motion.
And especially If simulation use the same code with the real robot then it definitely time efficient as well as evaluating the performance more accurately.

Lu-fm, 2021-01-01:
I have the same thought(

hijikata-toshi, 2021-04-08:
Excuse me! Is it literally impossible to test the real robot code in simulation environment?
I find that the '/kinova_bringup/kinova_robot.launch' will publish some topics like '/m1n6s200_driver/out' etc.,which are not publish in the 'kinova_gazebo/robot_launch.launch'. That makes it hard to directly transplant the real robotic code to simulation.
But is it possible to remap those real robot topic or modify the topic in the code, then test the code in gazebo?
Thank you!

closed 2021-06-07 by felixmaisonneuve