Kinovarobotics/kinova-ros #293: Is there any methond or API for motion control continuously?

HeegerGao, 2019-12-14:
If I use your Cartesian position control command:

rosrun kinova_demo pose_action_client.py -v -r m1n4s200 mdeg -- 0.01 0 0 0 10 10

I have to wait for about 1 second for finishing even when I just need a small movement. 

Is there any method that I could use for motion control continuously? I mean something like: I press a certain key and the robot will move to one certain direction until I release the key. I know how to get keyboard input, but I don't know which topic or server to call for this kind of function or how to use them properly. 

Thank you!

thommi1609, 2020-01-31:
I'm trying to do something similar, but am running into troubles as well.
My approach was to publish the pose on the topic "/_robottype_driver/pose_action/tool_pose/goal", with 20 Hz. 
I also edited the "kinova_tool_pose_action.cpp" so that the action_server also runs with 20 Hz (up from 10), and push the old goal as soon as a new one is received.
This is working to an extend, but the movement is jerky, and for some reason the jaco is overshooting quite a bit.
I have to admit i'm quite new to Ros, but i suspect this being an "action" slows things down?

I am also not very familiar with the kinowa_api.cpp and c++ in general, so I'd really like to avoid writing my own controller.

As @HeegerGao asked, isn't there an easily accessable way to control the Cartesian position continously?

Thank You!

polthergeist, 2020-03-20:
Hi @HeegerGao and @thommi1609 ,

One thing that you could check is to call directly the Kinova API to see if this solve your problem. Try the function:
`int SendBasicTrajectory(TrajectoryPoint trajectory);`
It is from the file Kinova.API.USBCommandLayerUbuntu.h
you can find the information about the trajectory point in the file: KinovaTypes.h
- Don't worry about the limitations.
- Don't worry about the SynchroType flag.
- Don't worry about the LimitationsActive flag.
Call the fucntion InitStruct() and you should good to set your cartesian position and send it.

Keep me posted,

HeegerGao, 2020-03-20:
> Hi @HeegerGao and @thommi1609 ,
>
> One thing that you could check is to call directly the Kinova API to see if this solve your problem. Try the function:
> `int SendBasicTrajectory(TrajectoryPoint [...]

I am now using Vrep and PyRep(which is a convenient library for robot arm control in both sim and real world environment) for my project. Maybe for people like me who only use jaco for high level research(for example, reinforcement learning) and not use it for fundemental robotics research, the low-level interface is not very attractive. Thank you anyway!

polthergeist, 2020-03-20:
Excellent! I will close this issue but feel free to reopen it if you have nay other question.

closed 2020-03-20 by HeegerGao