Kinovarobotics/kinova-ros #296: How to reduce the execution delay in ROS platform

sakthiprasasd, 2020-02-01:
Hi,
 The Kinova arm (J2s6s300) is showing a delay of one to two seconds when we are controlling through the ROS commands like " rosrun kinova_demo pose_action_client.py -v -r j2s6s300 mdeg -- 0.01 0 0 0 10 10". This kind of delay is not there when we control the arm using the Kinova development center(Graphical user interface) in Ubuntu. We tried with different systems but we are getting almost the same delay. 
Can you suggest any method to reduce the delay when we are controlling through the ROS (Indigo) commands? 
Should we need to modify any of the packages in the Kinova ROS to reduce this delay?

Simon-Steinmann, 2020-12-02:
Hi @sakthiprasasd,
controlling the robot using the pose_action_client.py script has to initialize and load many things, before it can execute anything. I suggest using rviz to controll the robot instead.

felixmaisonneuve, 2021-06-04:
You could also publish on different topics (depending on what you want to do) like this : 
```
rostopic pub -r 100 /j2n6s300_driver/in/joint_velocity kinova_msgs/JointVelocity "{joint1: 0.0, joint2: 0.0, joint3: 40.0, joint4: 0.0, joint5: 0.0, joint6: 0.0, joint7: 0.0}"
```
* this will move joint3 at a speed of 40 degres/s

Since there have been no response, I will assume this issue has been resolved and I will close it.

closed 2021-06-04 by felixmaisonneuve