Kinovarobotics/kinova-ros #297: Not able to install Kinova driver in jetson tx2

sakthiprasasd, 2020-02-01:
Hi,
     we are not able to install the Kinova driver in jetson tx2. We are using the Kinova J2s6s300 model, the error which we are getting and the specification of the hardware is given below. Please help us to solve this problem. 

error message while installing Kinova package in jetson tx2
This operation require root privilege.
rm: cannot remove '/usr/lib/*Kinova*.so': No such file or directory
dpkg: error processing archive KinovaAPI-6.0.2-amd64.deb (--install):
 package architecture (amd64) does not match system (arm64)
Errors were encountered while processing:
 KinovaAPI-6.0.2-amd64.deb

Jetson tx2 Specs:
 Features an integrated 256-core NVIDIA Pascal GPU, a hex-core ARMv8 64-bit CPU complex, and 8GB of LPDDR4 memory with a 128-bit interface. The CPU complex combines a dual-core NVIDIA Denver 2 alongside a quad-core ARM Cortex-A57

In Jetson tx2 Ubuntu 16.04 and ROS kinetic are using

cedriclmenard, 2020-02-27:
Unfortunately, Kinova does not provide binaries for ARM64. The ROS drivers actually link to shared libraries that are available only for x64_86 (aka 64 bits CISC CPUs like Intel's and AMD's) and i386 (32 bits CISC). No binaries for ARM and the source files for drivers are not available, so no way to actually compile the drivers on ARM Linux. 

As long as Kinova does not provide either the source or compile binaries for other architectures, this issue can't be solved.

sakthiprasasd, 2020-02-28:
Ok. Thanks for your reply.

LazyEngineerToBe, 2020-03-04:
You might try this.  https://github.com/Kinovarobotics/Extra_code_examples_Jaco2arm/tree/master/sdk%20compiled%20for%20tx2

The Jetson is an aarch64 architecture.

closed 2021-06-04 by felixmaisonneuve