Kinovarobotics/kinova-ros #302: "Trajectory command failed" appears and the real robot does not move

hgsyo76, 2020-04-17:
## Description
I'm trying to run the m1n6s200 with ROS. I was able to check the operation of Joint position control and Finger position control. However, every time I try to use Cartesian position control or moveit, I get the following warning message and the robot doesn't move.

[ WARN] [1587054718.495861155]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): LINE 201, Trajectory command failed

## Version
ROS distribution :melodoc

## Any other information
Looking forward to your answer!

alexvannobel, 2020-04-21:
Hi @hgsyo76 ,

Can you specify what commands you use to launch the nodes / launch files and what command you use to send the Cartesian position? Also, when launching those ndoes, do you see error messages in the console?

Thank you,

Alex

hgsyo76, 2020-04-23:
Thank you for replying!! @alexvannobel 

I just use the command  in README.md

First I launch this
`roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=m1n6s200`

And if I wanna control cartesian position, I use like this command  in other console.
`rosrun kinova_demo pose_action_client.py -v -r m1n6s200 mdeg -- 0.01 0 0 0 10 10`

But If I use this command  like this message will appear in the console that I did roslaunch .
`[ WARN] [1587054718.495861155]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): LINE 201, Trajectory command failed`

As for the moveit, I followed this.
https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_moveit/MoveIt.md

Thank you!!

alexvannobel, 2020-04-23:
Hi @hgsyo76 ,

Thanks for the details.
I think there are two elements that may help you here:
- First off, the robot won't execute cartesian commands if it wasn't homed prior to receiving the command. You can see how to to that with the dedicated ROS service call [here](https://github.com/Kinovarobotics/kinova-ros#ros-service-commands). You can also do it by holding the Home button on the Kinova Joystick.
- Secondly, is the pose you are sending valid? The translation elements `0.01 0 0` see very close to origin (base) to me. I suggest you can home the arm, then look at the arm's current Cartesian pose and adjust the values you send then (translate in one of the axis from the home position instead of `0 0 0`). It might be the source of `Trajectory command failed`.

Hope this helps,
Best,
Alex

yh0298, 2020-05-26:
Hello,

I met same error when I tried to joint control.

`rosrun kinova_demo joints_action_client j2n6s300 -v -r degree - 20 0 0 0 0 0`

and I found a solution on my case. 

jointSpeedLimitParameter1 was too low in 'robot_parameters.yaml' file. (ex. 0.5)

But I'm not sure why the speed limit affect on execution.

Is there minimum velocity for joint?

Looking forward to your answer!

alexvannobel, 2020-08-10:
Hi @yh0298 ,
Sorry for the long delay on this answer.

It is weird that your `jointSpeedLimitParameter1` was so low, it's higher in [the file we provide](https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_bringup/launch/config/robot_parameters.yaml#L5).

I don't know the exact value for the lower speed limit (`jointSpeedLimitParameter1` represents the **higher** speed limit for big actuators) but certainly having a too low higher speed limit messed with the trajectory execution.

I'll ask around to get the lower speed for the actuators and update you here, but I'll close this issue for now as the original issue was not updated following my answers and you seem to have resolved your issue.

Don't hesitate to reopen the issue or open a new one.
Best,
Alex

closed 2020-08-10 by alexvannobel

swords123, 2020-09-17:
Hello, I have also met the same problem, and in my robot_parameters.yaml, the jointSpeedLimitParameter1 is 10, what should I do?