Kinovarobotics/kinova-ros #303: C++11 Install Error

mfinean, 2020-05-02:
## Description

Clean catkin build in a new workspace results in errors from the kinova_driver package. Needs to be compiled with the C++11 additions

## Version

ROS distribution : Meloic

Branch and commit you are using : Master

## Steps to reproduce

catkin build

In file included from /home/user/kinova_ws/src/kinova-ros/kinova_driver/include/kinova_driver/kinova_ros_types.h:55:0,
                 from /home/user/kinova_ws/src/kinova-ros/kinova_driver/include/kinova_driver/kinova_comm.h:53,
                 from /home/user/kinova_ws/src/kinova-ros/kinova_driver/include/kinova_driver/kinova_joint_angles_action.h:54,
                 from /home/user/kinova_ws/src/kinova-ros/kinova_driver/src/kinova_joint_angles_action.cpp:47:
/opt/ros/melodic/include/tf/tf.h:354:8: error: ‘shared_ptr’ in namespace ‘std’ does not name a template type

## Proposed fix:
Add single line to the the CMakeLists.txt in the kinova_driver package:
add_compile_options(-std=c++11)

alexvannobel, 2020-05-11:
Hi @mfinean ,

That's weird, I too am on ROS Melodic and when I run `catkin build` on the tip of the master branch from a clean workspace, I manage to build without fail:
```
[build] Summary: All 17 packages succeeded!                                                                                                                                                                       
[build]   Ignored:   None.                                                                                                                                                                                        
[build]   Warnings:  5 packages succeeded with warnings.                                                                                                                                                          
[build]   Abandoned: None.                                                                                                                                                                                        
[build]   Failed:    None.                                                                                                                                                                                        
[build] Runtime: 1 minute and 58.3 seconds total.                                                                                                                                                                 
[build] Note: Workspace packages have changed, please re-source setup files to use them.
```
I understand the error and the quick fix, but I'd like to know why this error happens before pushing the fix. What is your version of `catkin`? What is your version of `gcc`?

Thanks,
Alex

mfinean, 2020-05-12:
Certainly: catkin_tools 0.4.5 and gcc 5.5.0

alexvannobel, 2020-05-12:
Thanks @mfinean 
I have gcc 7.5.0, so this could be because your gcc version is older that you get this error.
Anyway, it's a small fix so I made a PR.
I will let you know when it's merged.

closed 2020-08-10 by alexvannobel