Kinovarobotics/kinova-ros #305: Can not use moveit to control kinova arm in Gazebo

Hubert51, 2020-06-05:
## Description

I am trying to simulate the Kinova arm in the Gazebo and use moveit in rviz to control it. It seems I can not load the move_group part. The rviz was stuck there.

## Version

ROS distribution: kinetic

Branch and commit you are using : Master

## Steps to reproduce
I follow the step in the [Gazebo for Kinova robots
](https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_gazebo/Gazebo.md)
1. `roslaunch kinova_gazebo robot_launch.launch kinova_robotType:=j2n6s300`
2. `roslaunch j2n6s300_moveit_config j2n6s300_gazebo_demo.launch`

## The output in the terminal
```
PARAMETERS
 * /controller_list: [{'default': True...
 * /move_group/allow_trajectory_execution: True
 * /move_group/allowed_execution_duration_scaling: 1.2
 * /move_group/allowed_goal_duration_margin: 0.5
 * /move_group/arm/planner_configs: ['SBLkConfigDefau...
 * /move_group/capabilities: move_group/MoveGr...
 * /move_group/controller_list: [{'default': True...
 * /move_group/gripper/planner_configs: ['SBLkConfigDefau...
 * /move_group/jiggle_fraction: 0.05
 * /move_group/max_range: 5.0
 * /move_group/max_safe_path_cost: 1
 * /move_group/moveit_controller_manager: moveit_simple_con...
 * /move_group/moveit_manage_controllers: True
 * /move_group/octomap_resolution: 0.025
 * /move_group/planner_configs/BKPIECEkConfigDefault/border_fraction: 0.9
 * /move_group/planner_configs/BKPIECEkConfigDefault/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/BKPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/BKPIECEkConfigDefault/range: 0.0
 * /move_group/planner_configs/BKPIECEkConfigDefault/type: geometric::BKPIECE
 * /move_group/planner_configs/ESTkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/ESTkConfigDefault/range: 0.0
 * /move_group/planner_configs/ESTkConfigDefault/type: geometric::EST
 * /move_group/planner_configs/KPIECEkConfigDefault/border_fraction: 0.9
 * /move_group/planner_configs/KPIECEkConfigDefault/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/KPIECEkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/KPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/KPIECEkConfigDefault/range: 0.0
 * /move_group/planner_configs/KPIECEkConfigDefault/type: geometric::KPIECE
 * /move_group/planner_configs/LBKPIECEkConfigDefault/border_fraction: 0.9
 * /move_group/planner_configs/LBKPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/LBKPIECEkConfigDefault/range: 0.0
 * /move_group/planner_configs/LBKPIECEkConfigDefault/type: geometric::LBKPIECE
 * /move_group/planner_configs/PRMkConfigDefault/max_nearest_neighbors: 10
 * /move_group/planner_configs/PRMkConfigDefault/type: geometric::PRM
 * /move_group/planner_configs/PRMstarkConfigDefault/type: geometric::PRMstar
 * /move_group/planner_configs/RRTConnectkConfigDefault/range: 0.0
 * /move_group/planner_configs/RRTConnectkConfigDefault/type: geometric::RRTCon...
 * /move_group/planner_configs/RRTkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/RRTkConfigDefault/range: 0.0
 * /move_group/planner_configs/RRTkConfigDefault/type: geometric::RRT
 * /move_group/planner_configs/RRTstarkConfigDefault/delay_collision_checking: 1
 * /move_group/planner_configs/RRTstarkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/RRTstarkConfigDefault/range: 0.0
 * /move_group/planner_configs/RRTstarkConfigDefault/type: geometric::RRTstar
 * /move_group/planner_configs/SBLkConfigDefault/range: 0.0
 * /move_group/planner_configs/SBLkConfigDefault/type: geometric::SBL
 * /move_group/planner_configs/TRRTkConfigDefault/frountierNodeRatio: 0.1
 * /move_group/planner_configs/TRRTkConfigDefault/frountier_threshold: 0.0
 * /move_group/planner_configs/TRRTkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/TRRTkConfigDefault/init_temperature: 10e-6
 * /move_group/planner_configs/TRRTkConfigDefault/k_constant: 0.0
[... 86 lines cut ...]

WARNING: Package name "OARbot_description" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
process[move_group-1]: started with pid [12998]
process[rviz_ubuntu_12973_6669870134330938468-2]: started with pid [12999]
[ INFO] [1591335761.899671562]: Loading robot model 'j2n6s300'...
[ WARN] [1591335761.899752862]: Skipping virtual joint 'world_joint' because its child frame 'root' does not match the URDF frame 'world'
[ INFO] [1591335761.899768859]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1591335761.988857295]: rviz version 1.12.17
[ INFO] [1591335761.988912838]: compiled against Qt version 5.5.1
[ INFO] [1591335761.988929504]: compiled against OGRE version 1.9.0 (Ghadamon)
[ WARN] [1591335762.171644218, 1093.388000000]: Kinematics solver doesn't support #attempts anymore, but only a timeout.
Please remove the parameter '/robot_description_kinematics/arm/kinematics_solver_attempts' from your configuration.
pure virtual method called
terminate called without an active exception
[move_group-1] process has died [pid 12998, exit code -6, cmd /home/ruijie/kinova_ws/devel/lib/moveit_ros_move_group/move_group --debug /joint_states:=/j2n6s300/joint_states __name:=move_group __log:=/home/ruijie/.ros/log/c8f9433e-a6ec-11ea-bc90-48f17f8360f8/move_group-1.log].
log file: /home/ruijie/.ros/log/c8f9433e-a6ec-11ea-bc90-48f17f8360f8/move_group-1*.log
[ INFO] [1591335762.873164226, 1094.043000000]: Stereo is NOT SUPPORTED
[ INFO] [1591335762.873238061, 1094.043000000]: OpenGl version: 4.6 (GLSL 4.6).
[ INFO] [1591335766.238876813, 1097.369000000]: waitForService: Service [/get_planning_scene] has not been advertised, waiting...
[ WARN] [1591335771.307826872, 1102.388000000]: service '/get_planning_scene' not advertised yet. Continue waiting...
[ INFO] [1591335771.308260760, 1102.388000000]: waitForService: Service [/get_planning_scene] has not been advertised, waiting...

```

Thank you!

b-parikh, 2020-07-18:
Can you post the output of `rostopic list`?

Hubert51, 2020-07-31:
This is the output of `rostopic list`
```
/clock
/gazebo/link_states
/gazebo/model_states
/gazebo/set_link_state
/gazebo/set_model_state
/j2n6s300/effort_finger_trajectory_controller/command
/j2n6s300/effort_finger_trajectory_controller/follow_joint_trajectory/cancel
/j2n6s300/effort_finger_trajectory_controller/follow_joint_trajectory/feedback
/j2n6s300/effort_finger_trajectory_controller/follow_joint_trajectory/goal
/j2n6s300/effort_finger_trajectory_controller/follow_joint_trajectory/result
/j2n6s300/effort_finger_trajectory_controller/follow_joint_trajectory/status
/j2n6s300/effort_finger_trajectory_controller/gains/j2n6s300_joint_finger_1/parameter_descriptions
/j2n6s300/effort_finger_trajectory_controller/gains/j2n6s300_joint_finger_1/parameter_updates
/j2n6s300/effort_finger_trajectory_controller/gains/j2n6s300_joint_finger_2/parameter_descriptions
/j2n6s300/effort_finger_trajectory_controller/gains/j2n6s300_joint_finger_2/parameter_updates
/j2n6s300/effort_finger_trajectory_controller/gains/j2n6s300_joint_finger_3/parameter_descriptions
/j2n6s300/effort_finger_trajectory_controller/gains/j2n6s300_joint_finger_3/parameter_updates
/j2n6s300/effort_finger_trajectory_controller/state
/j2n6s300/effort_joint_trajectory_controller/command
/j2n6s300/effort_joint_trajectory_controller/follow_joint_trajectory/cancel
/j2n6s300/effort_joint_trajectory_controller/follow_joint_trajectory/feedback
/j2n6s300/effort_joint_trajectory_controller/follow_joint_trajectory/goal
/j2n6s300/effort_joint_trajectory_controller/follow_joint_trajectory/result
/j2n6s300/effort_joint_trajectory_controller/follow_joint_trajectory/status
/j2n6s300/effort_joint_trajectory_controller/gains/j2n6s300_joint_1/parameter_descriptions
/j2n6s300/effort_joint_trajectory_controller/gains/j2n6s300_joint_1/parameter_updates
/j2n6s300/effort_joint_trajectory_controller/gains/j2n6s300_joint_2/parameter_descriptions
/j2n6s300/effort_joint_trajectory_controller/gains/j2n6s300_joint_2/parameter_updates
/j2n6s300/effort_joint_trajectory_controller/gains/j2n6s300_joint_3/parameter_descriptions
/j2n6s300/effort_joint_trajectory_controller/gains/j2n6s300_joint_3/parameter_updates
/j2n6s300/effort_joint_trajectory_controller/gains/j2n6s300_joint_4/parameter_descriptions
/j2n6s300/effort_joint_trajectory_controller/gains/j2n6s300_joint_4/parameter_updates
/j2n6s300/effort_joint_trajectory_controller/gains/j2n6s300_joint_5/parameter_descriptions
/j2n6s300/effort_joint_trajectory_controller/gains/j2n6s300_joint_5/parameter_updates
/j2n6s300/effort_joint_trajectory_controller/gains/j2n6s300_joint_6/parameter_descriptions
/j2n6s300/effort_joint_trajectory_controller/gains/j2n6s300_joint_6/parameter_updates
/j2n6s300/effort_joint_trajectory_controller/state
/j2n6s300/joint_states
/rosout
/rosout_agg
/tf
/tf_static
```

b-parikh, 2020-08-01:
If I recall correctly, I faced something similar a while back. Give Moveit! some time to fully start up. You might see some errors that say certain things timed out. Then, before running `rosrun kinova_arm_moveit_demo pick_place`, run the command `rosrun kinova_gazebo get_current_state joint_states:=/j2n6s300/joint_states`. After you run this command, run the `pick_place` script. As a side note, can you interactively plan paths in Rviz?

alexvannobel, 2020-08-10:
Hi @Hubert51 ,

The program clearly exits because of `pure virtual method called`. 
A couple questions for you :
- Can you try following the MoveIt launch [instructions for a virtual robot](https://github.com/Kinovarobotics/kinova-ros/wiki/MoveIt#with-virtual-robot)?
- Are your ROS and MoveIt installations up-to-date and are all dependencies for `kinova-ros` installed?
- If the problem still persists, can you run the problematic node in gdb to get a backtrace of where the problem might happen? (see [this](https://answers.ros.org/question/47210/how-to-roslaunch-node-in-gdb/)). You're not launching any of our custom nodes so I'm guessing here the problem probably lies in the way the node is launched or in a configuration file (unless it's a problem in the third party-package).

Let me know how this goes,
Best,
Alex

felixmaisonneuve, 2021-06-04:
This issue has been inactive for a couple months now, I will close it.
If you still have this problem or encounter something related, feel free to reopen this issue or create a new one.

Thank you

closed 2021-06-04 by felixmaisonneuve