Kinovarobotics/kinova-ros #306: std::bad_alloc when calling moveit::planning_interface::MoveGroupInterface move_group("arm")

cellolegs, 2020-06-17:
I'm attempting to visualize some trajectories using the kinova j2n6s300 in MoveIt by running roslaunch j2s6s300_moveit_config demo.launch 
rosrun kinova_arm_moveit_demo motion_plan 

and keep getting the error std::bad_alloc at:

moveit::planning_interface::MoveGroupInterface move_group("arm");
in the kinova_arm_moveit_demo motion_plan.cpp line 31.

I've tried using the j2n6s300_moveit_config demo.launch as well as my own moveit config package and so far no luck. I really don't understand why I would have a memory issue (I have 16gb of ram and only firefox open) and I am able to use the same motion_plan code to run a different robot in moveit. I've tried the other kinova moveit config files and get the same issue. 

when I backtrace the issue in gdb I get:
[ INFO] [1592404981.586119869]: Loading robot model 'j2n6s300'...
[ WARN] [1592404981.588045462]: Skipping virtual joint 'world_joint' because its child frame 'root' does not match the URDF frame 'world'
[ INFO] [1592404981.588113068]: No root/virtual joint specified in SRDF. Assuming fixed joint
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc

Thread 1 "motion_plan" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff50e8801 in __GI_abort () at abort.c:79
#2  0x00007ffff5adb957 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff5ae1ae6 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff5ae1b21 in std::terminate() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff5ae1d54 in __cxa_throw () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff5ae22dc in operator new(unsigned long) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x00007ffff5b73b8b in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_mutate(unsigned long, unsigned long, char const*, unsigned long) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#8  0x00007ffff5b75133 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_append(char const*, unsigned long) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#9  0x00007ffff7626dd6 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > std::operator+<char, std::char_traits<char>, std::allocator<char> >(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
   from /opt/ros/melodic/lib/libmoveit_robot_model.so.1.0.2
#10 0x00007fffed24c720 in rospack::Rosstackage::getCachePath[abi:cxx11]() () from /opt/ros/melodic/lib/librospack.so
#11 0x00007fffed24cbe4 in rospack::Rosstackage::validateCache() () from /opt/ros/melodic/lib/librospack.so
#12 0x00007fffed253d78 in rospack::Rosstackage::readCache() () from /opt/ros/melodic/lib/librospack.so
#13 0x00007fffed253f25 in rospack::Rosstackage::crawl(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, bool) () from /opt/ros/melodic/lib/librospack.so
#14 0x00007fffed2631c4 in rospack::rospack_run(int, char**, rospack::Rosstackage&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&) () from /opt/ros/melodic/lib/librospack.so
#15 0x00007fffed25ee39 in rospack::ROSPack::run(int, char**) () from /opt/ros/melodic/lib/librospack.so
#16 0x00007fffed25f30b in rospack::ROSPack::run(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) () from /opt/ros/melodic/lib/librospack.so
#17 0x00007ffff0d9e761 in ros::package::command(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) () from /opt/ros/melodic/lib/libroslib.so
#18 0x00007ffff0d9e939 in ros::package::getPath(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) () from /opt/ros/melodic/lib/libroslib.so

Any ideas on how to resolve/investigate further? I have treid rebuilding the package in fresh workspace.

alexvannobel, 2020-08-10:
Hi @cellolegs ,
Sorry for the delay.
A couple questions for you:
- Are your ROS and MoveIt installations up-to-date (`apt update && apt upgrade`)? If you update a package, you should rebuild the workspace too.
- Can you use RViz to plan trajectories and move the arm with the MoveIt Motion Planning plugin? 
- What's the output of the `roslaunch j2s6s300_moveit_config demo.launch` terminal window whenever this error happens? Does it say anything special?

Let me know,
Best,
Alex

felixmaisonneuve, 2021-06-04:
This issue has been inactive for a couple months now, I will close it.
If you still have this problem or encounter something related, feel free to reopen this issue or create a new one.

Thank you

closed 2021-06-04 by felixmaisonneuve