Kinovarobotics/kinova-ros #308: "/j2n6s300_driver/in" topic not listed

hickeyc32, 2020-07-04:
Hello, 

I am using the Kinova simulation in gazebo and am trying to publish to the "/j2n6s300_driver/in/cartesian_velocity" and "/j2n6s300_driver/in/joint_velocity" topics as described in  **Velocity Control for joint space and Cartesian space**  of the README. However, these topics, or any topic with the "/j2n6s300_driver/" prefix, don't appear when I use rostopic list. Is there something I have to do to have /j2n6s300_driver/ appear?

Hubert51, 2020-08-03:
I think the simulation does not have these topics. These topics appear if you connect to a real arm.

alexvannobel, 2020-08-10:
Hi @hickeyc32 ,

As @Hubert51 pointed out, those topics are only available with a real arm.

You can achieve joint velocity control by using the individual position controllers and controlling them at a given rate with position increments. 
For Cartesian space velocity control though, there is no equivalent to the "/j2n6s300_driver/in/cartesian_velocity" topic.

Best,
Alex

closed 2020-08-10 by alexvannobel