Kinovarobotics/kinova-ros #310: How to bring finger movement into rviz

JorenFrans, 2020-07-09:
Hi everyone!

We are trying to implement a kinova-robotarm into rviz and make it possible to completely control it through this simulation-program. We almost got everything to work, just the fingers are still only controllable through code. We also can't find a way to implement them into rviz on the internet. Does anyone know if it's possible and how?

thanks in advance!
Joren

alexvannobel, 2020-08-10:
Hi @JorenFrans ,

Can you be more precise on what is asked here?
When you say "controllable through code", do you mean you want to use RViz to move the fingers instead of a node?
Also, the fingers are already implemented in RViz (they are attached to the arm). Do you mean movement in Gazebo cannot be seen from RViz?

Best,
Alex

felixmaisonneuve, 2021-06-04:
This issue has been inactive for a couple months now, I will close it.
If you still have this problem or encounter something related, feel free to reopen this issue or create a new one.

Thank you

closed 2021-06-04 by felixmaisonneuve