Kinovarobotics/kinova-ros #313: hwo to wire FT-300 and 2F-85(gripper) to Kinova Gen3 sucht that I can still control and read data from Kinova API or ROS?

JDchen-Jaden, 2020-08-28:
## Summary

Hello! I have RobotiQ FT-300 and a gripper (2F-85) need to be integrated onto the Kinova Gen3. They are not connected to the interface of Gen3 but can be connected to the computer via USB. Anyone know how to wire the FT-300 and gripper to the robot (I know there is one USB port on the BASE) such that I can control the gripper and read data from the sensor by Gen3 robot instead of the RobotiQ APPs.

If anyone has the same configuration as the picture I attached, could you please help me out? Thanks!

[image]

## Use case

Why is this feature needed?

What are the use case covered by this feature?

## Alternative

Do you have an idea how on how best to implement this enhancement?

If this feature is not implemented, do you have a workaround?

## Possible issues

Anything that could be affected by the new feature.

## Additional context

Any other information you would like us to know about the requested enhancement.

nberthelot-kino, 2020-08-28:
This is not supported. The arm cannot communicate with the FT300. All readings must be made via the usb cable connecting the FT300 to the Robotiq API on an external pc.

Thank you

JDchen-Jaden, 2020-08-30:
> This is not supported. The arm cannot communicate with the FT300. All readings must be made via the usb cable connecting the FT300 to the Robotiq API on an external pc.
> 
> Thank you

Thanks! I want to implement the admittance control by reading he FT 300 data, is there any way that I can log the data to the robot?

felixmaisonneuve, 2021-06-04:
Hi @JDchen-Jaden,

Do you still need help to implement admittance with your FT300 gripper?

felixmaisonneuve, 2021-07-26:
Hi @JDchen-Jaden,

I will close this issue. If you still have a problem, feel free to re-open or create a new issue.

Thank you,
Felix

closed 2021-07-26 by felixmaisonneuve