Kinovarobotics/kinova-ros #316: Attached Robot Arm exploding when launching gazebo together with the robot.

HappySamuel, 2020-09-18:
## Description
Loading the kinova j2s6s200 model with a robot when startup, but the robot arm will explode. 
[image]

## Version

ROS distribution : ROS Kinetic
Gazebo Version: Gazebo 9

Branch and commit you are using : master

## Steps to reproduce

1. 
```
<xacro:include filename="$(find kinova_description)/urdf/j2s6s200.xacro"/>
<xacro:j2s6s200 base_parent="base_link">
    <xacro:property name="joint_base_origin_xyz" value="0.317 -0.271 0.125"/>
    <xacro:property name="joint_base_origin_rpy" value="0 0 ${M_PI/2}"/>
</xacro:j2s6s200>
```

## Code example (if necessary)

```
<!-- Spawn Robot in Gazebo -->
<node pkg="gazebo_ros" type="spawn_model" name="spawn_model" output="screen"
	args="-urdf -model $(arg robot_namespace) -param robot_description -pause
		-x $(arg x)
		-y $(arg y)
		-z $(arg z)
		-Y $(arg yaw)
		-J $(arg kinova_robotType)_joint_1 4.712
		-J $(arg kinova_robotType)_joint_2 1.571
		-J $(arg kinova_robotType)_joint_3 0.523
		-J $(arg kinova_robotType)_joint_4 0.0
		-J $(arg kinova_robotType)_joint_5 4.0
		-J $(arg kinova_robotType)_joint_6 0.0
		-J $(arg kinova_robotType)_joint_finger_1 1.0
		-J $(arg kinova_robotType)_joint_finger_2 1.0"/>
```

```
<!-- Effort Joint trajectory controller-->
<node pkg="controller_manager" type="spawner" name="$(arg kinova_robotName)_trajectory_controller" output="screen"
	ns="$(arg kinova_robotName)" 
	args="effort_joint_trajectory_controller
	effort_finger_trajectory_controller    
	finger_tip_1_position_controller
	finger_tip_2_position_controller 
	joint_state_controller"/>   
```

## Expected behavior

Loading successfully without robot arm exploding.

## Any other information
How shall i solve this issue?

Best,
Samuel

HappySamuel, 2020-09-29:
Hi all,

I have found the root cause of causing this explosion of joints in Gazebo. The reason is due to the default PID value of robot arm that Kinova provided only applicable to default Gazebo physics settings on more complex robots.

Default Setting:
`    <physics name='default_physics' default='0' type='ode'>
      <max_step_size>0.001</max_step_size> 
      <real_time_factor>1</real_time_factor>
      <real_time_update_rate>1000</real_time_update_rate>
    </physics>
`

My Setting:
`    <physics name='default_physics' default='0' type='ode'>
      <max_step_size>0.005</max_step_size> 
      <real_time_factor>1</real_time_factor>
      <real_time_update_rate>200</real_time_update_rate>
    </physics>
`
Although my physics configuration will force the simulation to run at the `real_time_factor=1` , which is good especially when your laptop CPU is not the high end specs. With this method, you can still have the robot navigation working at real world speed. 
However, this bring disaster to the robotics arm simulation, the arm will explode if you attach onto another robot. 

According to my findout, if the max_step_size is larger than 0.001, the robot arm will start jittering. If the value go above 0.003, the finger will start to jittering apart from the joints.

Questions:
(1) Why is the max_step_size will affect the robot arm simulation?
(2) What shall i do to solve this exploding / jittering issue if want to apply my physics config (to ensure mobile robot + robot arm always simulated at real time speed.

Best,
Samuel

felixmaisonneuve, 2021-06-04:
Hi @HappySamuel, 
Do you still encounter a problem while using your custom physics settings?

I am not an expert in Gazebo, but I will be glad to help you in any way I can

Regards,
Felix

felixmaisonneuve, 2021-07-26:
Hi @HappySamuel,

I will close this issue. If you still have a problem, feel free to re-open or create a new issue.

Thank you,
Felix

closed 2021-07-26 by felixmaisonneuve

HappySamuel, 2021-07-27:
Hi @felixmaisonneuve 

Sorry for forgetting to close the issue down. I have already resolved the issue.

Best,
Samuel