Kinovarobotics/kinova-ros #318: Group gripper cannot work in Gazebo + Moveit

HappySamuel, 2020-10-05:
## Description
[image]

[image]

Error 1: Group 'gripper' is not a chain

Error 2: Kinematics solver of type 'kdl_kinematics_plugin/KDLKinematicsPlugin' could not be initialized for group 'gripper'

Error 3: Kinematics solver could not be instantiated for joint group gripper.

## Version

ROS distribution : ROS Kinetic + Gazebo 9.0

Branch and commit you are using : master

## Steps to reproduce

1. roslaunch kinova_gazebo robot_launch.launch kinova_robotType:=j2s6s300
2. roslaunch j2s6s300_moveit_config j2s6s300_gazebo_demo.launch

## Expected behavior
Gripper control available with moveit.

Need help/guidance on this, how to get the gripper controlled by moveit as well. I am following the tutorial, but found this error.

Best,
Samuel

soongjian, 2020-12-02:
I have the same problem. Have you solved it?

HappySamuel, 2020-12-03:
Hi @soongjian 

I yet to get it work. Hope @KinovaCustomerSupport can guide us on how to get it work.

Best,
Samuel

alexvannobel, 2020-12-08:
Hi @soongjian and @HappySamuel ,
Sorry for the long delay.
I'm looking into this right now.
I can reproduce the error and it also seems the j2s**7**s300 works fine on my end, so I'm looking into package-specific errors by comparing the two.
I'll keep you posted.
Alex

alexvannobel, 2020-12-08:
I've created a PR, it seems like the implementation differed because a kinematics solver was defined for the gripper, which does not make any sense and createds the error.
I was able to launch the simulation, send the gripper to the "Open" position and then to the "Close" position with the change.
The merge will close the issue but don't hesitate to reopen if you still experience problems.

Best,
Alex

PR #324 (2020-12-18): Fixed the gripper initialization problem
Fixes #318
closed 2020-12-18 by simonferronforget