Kinovarobotics/kinova-ros #320: Segmentation fault (core dumped) [gazebo-2] process has died [pid 4599, exit code 139

superggb, 2020-10-21:
## Description

I am using kinova_gazebo demo to run simulation of Mico_m1n6s300. I did not change the script too much for just added a sphere into the world.
It's same as the script below:
  roslaunch kinova_gazebo robot_launch.launch kinova_robotType:=m1n6s300
Then I use actionlib to control the arm in the world to do some RL training.
However it comes up to the  'Segmentation fault (core dumped) [gazebo-2] process has died' etc. randomly. Maybe after 500 epoch or maybe after 2000 epoch and so on.
I think it may be some problem inside gazebo. Is there any idea to solute this?

## Version

ROS distribution : ubuntu16.04 + ros kinetic

Branch and commit you are using : master

## Steps to reproduce

1. roslaunch kinova_gazebo robot_launch.launch kinova_robotType:=m1n6s300 (same as this)
2.  python train.py (My RL training script using actionlib to control arm)

## error message
Segmentation fault (core dumped)
[gazebo-2] process has died [pid 4599, exit code 139, cmd /opt/ros/kinetic/lib/gazebo_ros/gzserver -u -e ode /home/ggb/gym_ws/src/kinova-ros/kinova_gazebo/worlds/jaco.world __name:=gazebo __log:=/home/ggb/.ros/log/4d6fd7c0-1391-11eb-9d6e-64006a18b3f8/gazebo-2.log].
log file: /home/ggb/.ros/log/4d6fd7c0-1391-11eb-9d6e-64006a18b3f8/gazebo-2*.log

Same as this but pid may be different everytime

felixmaisonneuve, 2021-05-31:
Hi @superggb, 
Kinova provides a demo that moves the arm in kinova_control/src/move_robot.py.
You could try to run it to see if there is still an error. Just make sure you change the position to make the movement visible
```
    if (nbJoints==6):
      #home robots
      moveJoint ([0.0,2.9,1.3,4.2,1.4,0.0],prefix,nbJoints) # Change values
    else:
      moveJoint ([0.0,2.9,0.0,1.3,4.2,1.4,0.0],prefix,nbJoints)
```
You can run it with `python ${catkin_workspace}/src/kinova-ros/kinova_control/src/move_robot.py m1n6s300` or simply 
`rosrun kinova_control move_robot.py m1n6s300` when the Gazebo simulation is running

It might be a good idea to modify move_robot.py to make the arm move indefinitely to have a movement of more than just a couple seconds. 

If the error is also present, it will confirm your python script does not provoke the error and it probably comes from your Gazebo simulation like you suspect.
In that case, could you provide the modifications you did in your simulation?

If there is not an error anymore, the error is caused by your python script. In that case, can you provide your script?

Thank you,
Felix

felixmaisonneuve, 2021-07-26:
Hi @superggb ,

I will close this issue. If you still have a problem, feel free to re-open or create a new issue.

Thank you,
Felix

closed 2021-07-26 by felixmaisonneuve