Kinovarobotics/kinova-ros #321: Fail to roslaunch kinova_gazebo robot_launch.launch

Davincispencil, 2020-10-26:
I have just downloaded Kinova-ros from master and tried to launch kinova_gazebo, but I got this wired problem and could only open gazebo with empty world. 
[image]

I am currently using Ubuntu 16.04 LTS, Kinova-ros Kinetic-devel, Gazebo7, 

In addition, I can successfully open UR5 robot.

Could you please give me some advice?

For details:
yan@DaVin:~$ roslaunch kinova_gazebo robot_launch.launch 
... logging to /home/yan/.ros/log/1f52e666-178d-11eb-b67e-40b076a3ee78/roslaunch-DaVin-30301.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/loader.py:409: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  data = yaml.load(text)
started roslaunch server http://DaVin:38319/

SUMMARY
========

PARAMETERS
 * /j2n6s300/effort_finger_trajectory_controller/constraints/goal_time: 1.0
 * /j2n6s300/effort_finger_trajectory_controller/constraints/j2n6s300_joint_finger_1/goal: 0.02
 * /j2n6s300/effort_finger_trajectory_controller/constraints/j2n6s300_joint_finger_1/trajectory: 0.05
 * /j2n6s300/effort_finger_trajectory_controller/constraints/j2n6s300_joint_finger_2/goal: 0.02
 * /j2n6s300/effort_finger_trajectory_controller/constraints/j2n6s300_joint_finger_2/trajectory: 0.05
 * /j2n6s300/effort_finger_trajectory_controller/constraints/j2n6s300_joint_finger_3/goal: 0.02
 * /j2n6s300/effort_finger_trajectory_controller/constraints/j2n6s300_joint_finger_3/trajectory: 0.05
 * /j2n6s300/effort_finger_trajectory_controller/constraints/stopped_velocity_tolerance: 0.02
 * /j2n6s300/effort_finger_trajectory_controller/gains/j2n6s300_joint_finger_1/d: 0
 * /j2n6s300/effort_finger_trajectory_controller/gains/j2n6s300_joint_finger_1/i: 0
 * /j2n6s300/effort_finger_trajectory_controller/gains/j2n6s300_joint_finger_1/i_clamp: 1
 * /j2n6s300/effort_finger_trajectory_controller/gains/j2n6s300_joint_finger_1/p: 10
 * /j2n6s300/effort_finger_trajectory_controller/gains/j2n6s300_joint_finger_2/d: 0
 * /j2n6s300/effort_finger_trajectory_controller/gains/j2n6s300_joint_finger_2/i: 0
 * /j2n6s300/effort_finger_trajectory_controller/gains/j2n6s300_joint_finger_2/i_clamp: 1
 * /j2n6s300/effort_finger_trajectory_controller/gains/j2n6s300_joint_finger_2/p: 10
 * /j2n6s300/effort_finger_trajectory_controller/gains/j2n6s300_joint_finger_3/d: 0
 * /j2n6s300/effort_finger_trajectory_controller/gains/j2n6s300_joint_finger_3/i: 0
 * /j2n6s300/effort_finger_trajectory_controller/gains/j2n6s300_joint_finger_3/i_clamp: 1
 * /j2n6s300/effort_finger_trajectory_controller/gains/j2n6s300_joint_finger_3/p: 10
 * /j2n6s300/effort_finger_trajectory_controller/joints: ['j2n6s300_joint_...
 * /j2n6s300/effort_finger_trajectory_controller/type: effort_controller...
 * /j2n6s300/effort_joint_trajectory_controller/constraints/goal_time: 1.0
 * /j2n6s300/effort_joint_trajectory_controller/constraints/j2n6s300_joint_1/goal: 0.02
 * /j2n6s300/effort_joint_trajectory_controller/constraints/j2n6s300_joint_1/trajectory: 0.05
 * /j2n6s300/effort_joint_trajectory_controller/constraints/j2n6s300_joint_2/goal: 0.02
 * /j2n6s300/effort_joint_trajectory_controller/constraints/j2n6s300_joint_2/trajectory: 0.05
 * /j2n6s300/effort_joint_trajectory_controller/constraints/j2n6s300_joint_3/goal: 0.02
 * /j2n6s300/effort_joint_trajectory_controller/constraints/j2n6s300_joint_3/trajectory: 0.05
 * /j2n6s300/effort_joint_trajectory_controller/constraints/j2n6s300_joint_4/goal: 0.02
 * /j2n6s300/effort_joint_trajectory_controller/constraints/j2n6s300_joint_4/trajectory: 0.05
 * /j2n6s300/effort_joint_trajectory_controller/constraints/j2n6s300_joint_5/goal: 0.02
 * /j2n6s300/effort_joint_trajectory_controller/constraints/j2n6s300_joint_5/trajectory: 0.05
 * /j2n6s300/effort_joint_trajectory_controller/constraints/j2n6s300_joint_6/goal: 0.02
 * /j2n6s300/effort_joint_trajectory_controller/constraints/j2n6s300_joint_6/trajectory: 0.05
 * /j2n6s300/effort_joint_trajectory_controller/constraints/stopped_velocity_tolerance: 0.02
 * /j2n6s300/effort_joint_trajectory_controller/gains/j2n6s300_joint_1/d: 0
 * /j2n6s300/effort_joint_trajectory_controller/gains/j2n6s300_joint_1/i: 0
 * /j2n6s300/effort_joint_trajectory_controller/gains/j2n6s300_joint_1/i_clamp: 10
 * /j2n6s300/effort_joint_trajectory_controller/gains/j2n6s300_joint_1/p: 5000
 * /j2n6s300/effort_joint_trajectory_controller/gains/j2n6s300_joint_2/d: 0
 * /j2n6s300/effort_joint_trajectory_controller/gains/j2n6s300_joint_2/i: 0
 * /j2n6s300/effort_joint_trajectory_controller/gains/j2n6s300_joint_2/i_clamp: 10
 * /j2n6s300/effort_joint_trajectory_controller/gains/j2n6s300_joint_2/p: 5000
 * /j2n6s300/effort_joint_trajectory_controller/gains/j2n6s300_joint_3/d: 0
 * /j2n6s300/effort_joint_trajectory_controller/gains/j2n6s300_joint_3/i: 0
[... 79 lines cut ...]

process[gazebo-1]: started with pid [30321]
process[gazebo_gui-2]: started with pid [30325]
process[urdf_spawner-3]: started with pid [30330]
process[j2n6s300/j2n6s300_trajectory_controller-4]: started with pid [30332]
process[robot_state_publisher-5]: started with pid [30333]
process[command_robot_home_pose-6]: started with pid [30334]
[ERROR] [1603718180.077096158]: Parser found 1 elements but 3 expected while parsing vector [-0.10117000000000001]
[ERROR] [1603718180.077164561]: Malformed parent origin element for joint [j2n6s300_joint_5]
[ERROR] [1603718180.077178316]: joint xml is not initialized correctly
[ INFO] [1603718180.259177876]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1603718180.260081507]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[robot_state_publisher-5] process has died [pid 30333, exit code 255, cmd /opt/ros/kinetic/lib/robot_state_publisher/robot_state_publisher /joint_states:=/j2n6s300/joint_states __name:=robot_state_publisher __log:=/home/yan/.ros/log/1f52e666-178d-11eb-b67e-40b076a3ee78/robot_state_publisher-5.log].
log file: /home/yan/.ros/log/1f52e666-178d-11eb-b67e-40b076a3ee78/robot_state_publisher-5*.log
[ INFO] [1603718180.275771761]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1603718180.277144293]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
SpawnModel script started
[INFO] [1603718180.447214, 0.000000]: Controller Spawner: Waiting for service controller_manager/load_controller
[INFO] [1603718180.563747, 0.000000]: Loading model XML from ros parameter
[INFO] [1603718180.567316, 0.000000]: Waiting for service /gazebo/spawn_urdf_model
[ INFO] [1603718181.456454918]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1603718181.458461057]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[INFO] [1603718181.473849, 0.000000]: Calling service /gazebo/spawn_urdf_model
Error [parser_urdf.cc:3474] Unable to call parseURDF on robot model
Error [parser.cc:341] parse as old deprecated model file failed.

kashifzr, 2021-01-11:
[ERROR] [1603718180.077096158]: Parser found 1 elements but 3 expected while parsing vector [-0.10117000000000001]
[ERROR] [1603718180.077164561]: Malformed parent origin element for joint [j2n6s300_joint_5]
[ERROR] [1603718180.077178316]: joint xml is not initialized correctly

try to resolve these issue otherwise it will not

felixmaisonneuve, 2021-05-31:
Hi @kashifzr,
You are saying you just downloaded the kinova repo and launched Gazebo with
`~$ roslaunch kinova_gazebo robot_launch.launch` and you get these errors?
Have you verified if all the required packages are installed and up to date?
```
sudo apt-get install ros-indigo-gazebo-ros*  
sudo apt-get install ros-<distro>-gazebo-ros-control
sudo apt-get install ros-<distro>-ros-controllers*
sudo apt-get install ros-<distro>-trac-ik-kinematics-plugin
```

Also, could you please edit your comment and add trible backticks at the beginning and end of your terminal output to put everything in a code block for better scrolling and better display.

You can learn more on markdown syntax here : https://guides.github.com/features/mastering-markdown/

Regards,
Felix

closed 2021-06-11 by Davincispencil