Kinovarobotics/kinova-ros #323: Plam is upside down in urdf of j2n6s200

flymaxty, 2020-11-19:
Hi,

When the j2n6s200 is at the Home position, the plam in Rviz is different with real robot, it is upside down.
I know that might not a big issue but, the palm is actually asymmetrical.
Is there anyway to adjust it?

Thanks

felixmaisonneuve, 2021-06-01:
Hi @flymaxty,
I have a j2n6s300 and my palm is also inverted. You can change the orientation in kinova_description/urdf/j2n6s200. 
You should have a line like `<xacro:property name="joint_6_origin_rpy" value="${J_PI/3} 0 ${J_PI}" />`
you need to change the values to `value="${-J_PI/3} 0 0"`

I feel like that change should have been in `joint_end_effector_origin_rpy`, but it doesn't seem to change anything...
Also, I have looked at all the closed issues and I didn't see any mention to this, which seems odd after all these years for a problem this obvious. 

Anyway, this change should do the trick

Regards,
Felix

closed 2021-07-26 by felixmaisonneuve

z111222333, 2024-05-15:
Hello, why did I switch from the xacro file to the urdf file? There is no robot configuration-related content in it, which is not what a urdf should have.