Kinovarobotics/kinova-ros #325: Control trajectory for 1 joint

Jelvani, 2020-12-12:
I am trying to use a j2n6s200 arm in a real-time manner, needing to control the position in a control loop with other sensors. Originally, I tried to use the actionserver to send position commands, but this was not ideal, since I need to following a trajectory rather, since the positions to go to are coming in at a constant rate. 
So I am now trying to use the `joint_trajectory/command` topic to publish a trajectory, hoping to just test a sine wave. I am having some issues, namely, when I publish the trajectory message, all joints are moving, rather than just the one I specified in the array of positions.
My question now is, what is the correct way to specify a trajectory to this arm? Because, I cannot put multiple points in a trajectory message, I am sending 1 point at a time. So this means I am publishing the trajectory messages at around 100hz. 
Is there a way I can control the trajectory in real-time for 1 joint?

alexvannobel, 2021-01-06:
Hi @Jelvani ,
Sorry for the long delay.

You can try joint velocity commands. You can have a look at [this section](https://github.com/Kinovarobotics/kinova-ros#velocity-control-for-joint-space-and-cartesian-space) in the documentation. The control **has to be** 100 Hz and this way you could be able to control position increments to a single joint.

I think this is the easiest way to control only one joint at a time while maintaining the other ones static. I'll have access to an arm later this week so I'll have a look at all the topics to see if there is a better solution, but for now this is my advice.

Hope this helps, 
Best regards,
Alex

Jelvani, 2021-01-06:
> Hi @Jelvani ,
>
> Sorry for the long delay.
>
>
>
> You can try joint velocity commands. You can have a look at [this [...]

Thank you for the reply, the problem is, I would need feedback from the arm position, and the default feedback rate is 10hz. Increasing this to 100hz, I see the maximum won't go above 40hz, and I am also not able to then control the arm with trajectory or velocity commands. I assume this is some bandwidth issue with the microcontroller used in the arm. I currently am stuck here.

alexvannobel, 2021-01-06:
Hi @Jelvani ,

Do you only need the joint angles feedback from the driver?

If so, you could try [commenting the publishing of data you don't need](https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_driver/src/kinova_arm.cpp#L733-L735) in the driver and rebuild it.

The arm's API effectively functions at a rate of 100Hz, so reducing the number of useless calls made by the driver to the arm should get you better performance on the feedback.

Tell me how this goes,
Best regards,
Alex

closed 2021-01-14 by Jelvani