Kinovarobotics/kinova-ros #326: Joint velocity control

QuarterByte, 2020-12-17:
Not able to control the joint velocity of j2s6s300 with python program. when using  #rostopic pub .....# , it is working but when I tried the same topic with my python program, it didn't work.

felixmaisonneuve, 2021-05-31:
Hi @QuarterByte,
can you provide more information about this issue?
You say it doesn't work with a python program. Is it your program or are you trying to use joints_action_client.py which is provided in the kinova_demo packages?

Also, if it is your own program, can you write here so I can better understand what the issue is?

Thank you,
Felix

felixmaisonneuve, 2021-07-26:
Hi @QuarterByte ,

I will close this issue. If you still have a problem, feel free to re-open or create a new issue.

Thank you,
Felix

closed 2021-07-26 by felixmaisonneuve