Kinovarobotics/kinova-ros #327: Joint trajectory control does not seem to work

Jelvani, 2021-01-18:
I have the following python ROS node below, publishing a joint trajectory message to `/j2n6s200_driver/trajectory_controller/command`. It should be moving only joint 5 on the arm, but for some reason joints 1,2 and 3 are moving to random positions, the single joint I command doesn't even move. Can somebody please show how you are supposed to publish a joint trajectory message for angular position control?

```python
#!/usr/bin/env python3
import rospy, math, time, numpy
from std_msgs.msg import Header
import kinova_msgs.msg, actionlib
from trajectory_msgs.msg import JointTrajectory
from trajectory_msgs.msg import JointTrajectoryPoint

PI = math.pi
	
def home(angles = [180,180,180,180,180,180]):
	action_address = '/j2n6s200_driver/joints_action/joint_angles'
	client = actionlib.SimpleActionClient(action_address,kinova_msgs.msg.ArmJointAnglesAction)
	client.wait_for_server()
	goal = kinova_msgs.msg.ArmJointAnglesGoal()
	goal.angles.joint1 = angles[0]
	goal.angles.joint2 = angles[1]
	goal.angles.joint3 = angles[2]
	goal.angles.joint4 = angles[3]
	goal.angles.joint5 = angles[4]
	goal.angles.joint6 = angles[5]
	client.send_goal(goal)
	client.wait_for_result()

def moveJoint (jointcmds,prefix,nbJoints):
  topic = '/j2n6s200_driver/trajectory_controller/command'
  pub = rospy.Publisher(topic, JointTrajectory, queue_size=1)
  jointCmd = JointTrajectory()  
  point = JointTrajectoryPoint()
  jointCmd.header.stamp = rospy.Time.now() + rospy.Duration.from_sec(0.0);  
  point.time_from_start = rospy.Duration.from_sec(5.0)
  for i in range(0, nbJoints):
    jointCmd.joint_names.append(prefix +'_joint_'+str(i+1))
    point.positions.append(jointcmds[i])
    point.velocities.append(0)
    point.accelerations.append(0)
    point.effort.append(0) 
  jointCmd.points.append(point)
  rate = rospy.Rate(100)
  count = 0
  while (count < 50):
    pub.publish(jointCmd)
    count = count + 1
    rate.sleep() 

if __name__=='__main__':
	rospy.init_node('arm_traj')
	home()
	rate=rospy.Rate(100)
	moveJoint([PI,PI,PI,PI,PI/2,PI],'j2n6s200',6)
```

alexvannobel, 2021-01-29:
Hi @Jelvani ,

When trying your code with my j2n6s300, the `home` function does work properly but the `moveJoint` function does not make the robot move at all, but I believe you are actually supposed to provide a trajectory (and not a single point) in this publish, alongside correct `time_from_start` values for every point in your trajectory. It also would be better to use the an action here to know whether or not the trajectory is finished instead of blindly waiting in a rate.

In short, why not call the action server in a for loop for every point you want to reach in your trajectory?

Jelvani, 2021-01-30:
> Hi @Jelvani ,
>
>
>
> When trying your code with my j2n6s300, the `home` function does work properly but the `moveJoint` function does not make the robot move at all, but I believe you are actually [...]

The problem with the action server is that the position control means it goes to each point and stops, we also don't have velocity control at each point in the trajectory. With the above code, I have tried it with more points as well, but the trajectory control still does not work, I get the same results.
From what I understood, a 1 point trajectory would mean it goes from the current position, to the next point, in the given time for that point and given velocity. So 1 point would mean it goes from the current position, to the given position in the point, and stops since a velocity of 0 is given. 

Is it possible you could do a simple 2-3 point trajectory on your arm that works, and post the code? I am wondering if there is something else that is the issue here.

alexvannobel, 2021-02-03:
Hi @Jelvani ,

I did rework a bit your code to make the trajectory controller work, but there are a couple limitations.
1. I found out to my surprise that the `rospy.Publisher` object had to be create **before** the `rospy.init_node()` call, else the message would not be published, hence the `class` below.
2. The `/*_driver/trajectory_controller/command` 's callback implementation is found [in this function](https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_driver/src/kinova_joint_trajectory_controller.cpp#L81), and you can see that the "continuous position control" is actually achieved through velocity control, because the Jaco2 arm doesn't offer multi-point trajectories (without stopping), hence the velocity control through `in/joint_velocity`. Depending on your application and the level of precision you need, you might want to do velocity control yourself with that topic instead of using the `trajectory_controller/command` topic.
3. I changed the `moveJoint` to instead take a trajectory as input, and hardcoded the `time_from_start` as well as the velocities in the method. Obviously, for your purpose you should input variable `time_from_start` for each position. For the velocities, I easily hardcoded it here to rotate the end effector at pi/6 rad/s but they should in fact be calculated based on the previous point and also on the duration needed to reach the point. 
4. My robot is a j2n6s**3**00 so you might have to change that for your robot type to correctly run the example.

The robot goes to vertical position, then the end-effector rotates 180 degrees.

This is the code : 
```python
#!/usr/bin/env python3
import rospy
import math
import kinova_msgs.msg
import actionlib
from trajectory_msgs.msg import JointTrajectory, JointTrajectoryPoint

PI = math.pi

class Example:
    def __init__(self):
        trajectoryControllerTopic = '/j2n6s300_driver/trajectory_controller/command'
        self.pub = rospy.Publisher(trajectoryControllerTopic, JointTrajectory, queue_size=1)

    def goto(self, angles):
        action_address = '/j2n6s300_driver/joints_action/joint_angles'
        client = actionlib.SimpleActionClient(
            action_address, kinova_msgs.msg.ArmJointAnglesAction)
        client.wait_for_server()
        goal = kinova_msgs.msg.ArmJointAnglesGoal()
        goal.angles.joint1 = angles[0]
        goal.angles.joint2 = angles[1]
        goal.angles.joint3 = angles[2]
        goal.angles.joint4 = angles[3]
        goal.angles.joint5 = angles[4]
        goal.angles.joint6 = angles[5]
        client.send_goal(goal)
        client.wait_for_result()

    def trajControl(self, trajectory, prefix, nbJoints):
        jointCmd = JointTrajectory()
        for i in range(0, nbJoints):
            jointCmd.joint_names.append(prefix + '_joint_'+str(i+1))
        jointCmd.header.stamp = rospy.Time.now() + rospy.Duration.from_sec(0.0)
        for pointIndex in range(1, len(trajectory)):
            jointcmds = trajectory[pointIndex]
            point = JointTrajectoryPoint()
            point.time_from_start = pointIndex * rospy.Duration.from_sec(1.0)
            for i in range(0, nbJoints):
                point.positions.append(jointcmds[i])
                # velCmd is hardcoded but would need to be calculated based on previous point in traj
                velCmd = -PI/6 if (i == nbJoints - 1) else 0
                point.velocities.append(velCmd)
                # accelerations and efforts are unused
                # point.accelerations.append(0)
                # point.effort.append(0)
            jointCmd.points.append(point)
        self.pub.publish(jointCmd)

if __name__ == '__main__':
    e = Example()
    rospy.init_node('arm_traj')
    e.goto([180, 180, 180, 180, 180, 180])
    trajectory = [
        [PI, PI, PI, PI, PI, PI],
        [PI, PI, PI, PI, PI, 5*PI/6],
        [PI, PI, PI, PI, PI, 2*PI/3],
        [PI, PI, PI, PI, PI, PI/2],
        [PI, PI, PI, PI, PI, PI/3],
        [PI, PI, PI, PI, PI, PI/6],
        [PI, PI, PI, PI, PI, 0],
    ]
    e.trajControl(trajectory, 'j2n6s300', 6)
```

Hope that helps,
Alex

Jelvani, 2021-02-03:
@alexvannobel ,
Thank you so much for your help and time, 
this information is really helpful to know for us trying to do trajectory control, along with the class you provided.

closed 2021-02-03 by Jelvani