Kinovarobotics/kinova-ros #328: Position control demo with python in Gazebo

baifanxxx, 2021-01-28:
Hi author,
Your kinova repo is very useful and helps my research. However, I want to do simulation in gazebo with kinova, there is nothing python demo about controling the postion and grasp on kinova in gazebo. Can you share me some demos about controling the postion and grasp on kinova in gazebo? Thank you very much.

Best regards,
BAI Fan

felixmaisonneuve, 2021-05-31:
Hi @baifanxxx,
Informations about kinova robots on Gazebo can be found here : https://github.com/Kinovarobotics/kinova-ros/wiki/Gazebo

You are talking about demo. In the repo, you can find kinova_control/src/move_robot.py that can be used to move the robot withing the Gazebo simulation. This file is an example that shows how you can position each joint and each finger.

Just be aware that the default position in move_robot.py is the home position, which is the default position in Gazebo when you launch the simulation. So if you run the file without changing the values, the arm won't move. you'll want to change this : 
```
if (nbJoints==6):
      #home robots
      moveJoint ([0.0,2.9,1.3,4.2,1.4,0.0],prefix,nbJoints)
else:
      moveJoint ([0.0,2.9,0.0,1.3,4.2,1.4,0.0],prefix,nbJoints)
```
into something like this (for example) : 
```
if (nbJoints==6):
      #home robots
      moveJoint ([1.0,2.5,1.8,3.5,1.6,1.0],prefix,nbJoints)
else:
      moveJoint ([1.0,2.5,0.0,1.8,3.5,1.6,1.0],prefix,nbJoints)
```

same logic applies for the function `moveFingers` to change your grasps.

Hope this helps

felixmaisonneuve, 2021-07-26:
Hi @baifanxxx ,

I will assume your question was answered and close this issue. If you still have a problem, feel free to re-open or create a new issue.

Thank you,
Felix

closed 2021-07-26 by felixmaisonneuve