Kinovarobotics/kinova-ros #329: Real-Time Joint angle control

sirslab, 2021-03-16:
## Description

We are trying to control in real-time the position (i.e., the joint angle) of a single joint actuator. We are using a Mico 6 DoFs 2 fingers robot. To reduce problems related to the dynamics and load we are testing the last joint (joint6). 

We have created a simple node that publishes at 100Hz the desired joint value to `'/m1n6s200_driver/joints_action/joint_angles/goal'`.
We tested a simple sine reference signal, but we noticed that the behavior of the robot is unpredictable, meaning that sometimes the robot is stuck for seconds, some other times the speed is definitively not constant. More generally, the tracking error varies a lot from one trial to another, even if the reference trajectory is always the same.

We tried without success the following solutions/workarounds:
- publish a new goal no matter of the robot position
- publish a new goal only if the previous one is reached
- publish a new goal only if it is different from the previous one
- change the node rate (from 5Hz to 100Hz) 

here the main code code of the node

```
topic_name_pub = '/m1n6s200_driver/joints_action/joint_angles/goal'
pub_joints = rospy.Publisher(topic_name_pub, kinova_msgs.msg.ArmJointAnglesActionGoal, queue_size=10)
while not rospy.is_shutdown():
	pub_joints.publish(goal)
        rate.sleep()
```

Is there a way to modify the joint angle PID controller parameters? Alternatively, there exists a simpler way for precisely control the joint angle?

## Version

ROS distribution : 
Melodic
Branch and commit you are using : 
Master branch

felixmaisonneuve, 2021-05-31:
Hi @sirslab,
Assuming I understand correctly what you are doing and what you are trying to do,
You are sending an action at a frequency of 100Hz. Actions are supposed to be sent once, then the arm will move to this position automatically. 
What you want to do is publish a velocity (in your case joint_velocity). This will allow you to control the speed of each joint. 
In the command line it will look something like this (notice the `-r 100` to specify publish rate):
```~$ rostopic pub -r 100 /j2n6s300_driver/in/joint_velocity kinova_msgs/JointVelocity 0.0 0.0 0.0 20.0 0.0 0.0 0.0```
This command will publish at a rate of 100Hz (mandatory, otherwise the arm won't move as expected) a speed for each joint. The arm will then move each joint indefinitely at the requested speed. CTRL+c to stop.

It is also possible you want to send the arm to a specific position, but then you want to send an action once.
In the command line, it will look something like this (notice there is no `-r 100`) :
```
~$ rostopic pub /j2n6s300_driver/fingers_action/finger_positions/goal kinova_msgs/SetFingersPositionActionGoal "header:
  seq: 0
  stamp:
    secs: 0
    nsecs: 0
  frame_id: ''
goal_id:
  stamp:
    secs: 0
    nsecs: 0
  id: ''
goal:
  fingers:
    finger1: 500.0
    finger2: 500.0
    finger3: 500.0" 
publishing and latching message. Press ctrl-C to terminate
^C~$
```
This command will send a finger position once and the arm will move until it reaches this position. Publishing this command continuously before the action is complete will cause the arm to move sporadically or not at all (like what you saw)

I would recommand to use rqt. (just write `$ rqt` when the robot is launched) It provides a somewhat nice interface that shows you all the topics available with ROS and allows you to publish on different topics so you can quickly test everything you want.

I hope this makes sense and everything is clear
If anything, I'll be glad to help you

Regards, 
Felix

felixmaisonneuve, 2021-07-26:
Hi @sirslab ,

I will close this issue. If you still have a problem, feel free to re-open or create a new issue.

Thank you,
Felix

closed 2021-07-26 by felixmaisonneuve

JaeYoungLee79, 2021-12-03:
Where is the example?

Joint1 = 150
Joint2 = 200
....

felixmaisonneuve, 2021-12-03:
Hi @JaeYoungLee79,

I am not sure what example you are talking about, but you can find many informations in the readme file, more preciselyin the [joint position control](https://github.com/Kinovarobotics/kinova-ros#joint-position-control) section and the [cartesian position control](https://github.com/Kinovarobotics/kinova-ros#cartesian-position-control) section. 

Best,
Felix

JaeYoungLee79, 2022-09-29:
Dear Sir
I findout the example of joint control : 102-Movement_high_level
Thanks a lot

felixmaisonneuve, 2022-09-29:
Be careful if you are talking about these examples : https://github.com/Kinovarobotics/kortex/tree/master/api_cpp/examples/102-High_level_movement

This is from our Kortex repo, which is for our Gen3 arms. The examples **will not** work for your Jaco2 arm