Kinovarobotics/kinova-ros #331: angles of torque zero mode is not correct in gravity_compensated_mode.py

StanleyYake, 2021-03-17:
Hello there, I found angles of torque zero mode is not correct in gravity_compensated_mode.py in KINOVA MAINTENANCE GUIDE.pdf which refer to [180,180,180,180, 0, 90] candle like position, but the gravity_compensated_mode.py refers to [180, 180,180,180,180,180] 
[image]

https://github.com/Kinovarobotics/kinova-ros/blob/38293ed73bc230e9ca7e7312fc268736821fd9cc/kinova_demo/nodes/kinova_demo/gravity_compensated_mode.py#L32

YangSai567, 2021-04-12:
Hi, I found that as same as you.First. I only have JACO SDK, should that be able to reset torque control mode? Then, do you have ideals of starting gravity compensated mode? Thanks!

felixmaisonneuve, 2021-05-31:
Hi @StanleyYake,
Effectively, angles differ. 
However, gravity_compensated_mode.py does not reset the torque sensors, which means values could be anything. The objective being only to show what happens when all torques values are set to 0. 
Also, spherical and non-spherical wrist to not operate the same way, it is possible (not sure, I haven't verified) that 180 instead of 0 is a candle-like position for the spherical wrist. It might also be a typo, but then again, it doesn't really change anything. 

To answer to @YangSai567, yes, you can reset torque control settings in JACO-SDK.  You can access actuator angles in Monitoring section and you can set zero torque sensor in Advanced Settings. I am not sure what you mean by "ideals of starting gravity compensated mode".

felixmaisonneuve, 2021-06-01:
Also @StanleyYake, where did you find the maintenance guide you mentionned? I am not sure where this screenshot comes from. 
The user guide for Jaco2 available on Kinova's website recommend these values :
[image]

I also had confirmation that the arm position when reseting torque is not very important. It is recommended to put the arm in a candle-like position to reduce the effect on gravity on the sensors. Reseting the arm while it is in a horizontal position might affect calibration because of cross-torque (caused by gravity) on the joints, but it is possible to do so.

felixmaisonneuve, 2021-07-26:
Hi @StanleyYake,

I will assume your question was answered and close this issue. If you still have a problem, feel free to re-open or create a new issue.

Thank you,
Felix

closed 2021-07-26 by felixmaisonneuve