Kinovarobotics/kinova-ros #332: motion_plan.cpp no longers works with melodic

mpatalberta, 2021-03-31:
## Description
all plan and goal rejected (except when you go to the place you currently are at)

[ INFO] [1617220833.755000670]: Loading robot model 'j2s6s300'...
[ WARN] [1617220833.755043464]: Skipping virtual joint 'world_to_base_virtual_joint' because its child frame 'root' does not match the URDF frame 'world'
[ INFO] [1617220833.755065878]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ WARN] [1617220833.904444238]: Kinematics solver doesn't support #attempts anymore, but only a timeout.
Please remove the parameter '/rviz_tnl_26403_8251536395038140518/arm/kinematics_solver_attempts' from your configuration.
[ WARN] [1617220833.909101227]: IK plugin for group 'arm' relies on deprecated API. Please implement initialize(RobotModel, ...).
[ INFO] [1617220834.263082206]: Starting planning scene monitor
[ INFO] [1617220834.264822043]: Listening to '/move_group/monitored_planning_scene'
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
[ INFO] [1617220834.450175288]: Constructing new MoveGroup connection for group 'arm' in namespace ''
[ INFO] [1617220835.476527527]: Ready to take commands for planning group arm.
[ INFO] [1617220835.476665757]: Looking around: no
[ INFO] [1617220835.476733199]: Replanning: no
[ INFO] [1617220852.177467991]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[ INFO] [1617220852.178570474]: Planner configuration 'arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1617220852.178885809]: RRTConnect: Starting planning with 1 states already in datastructure
[ERROR] [1617220852.481327597]: RRTConnect: Unable to sample any valid states for goal tree
[ INFO] [1617220852.481396504]: RRTConnect: Created 1 states (1 start + 0 goal)
[ INFO] [1617220852.481421099]: No solution found after 0.302641 seconds
[ INFO] [1617220852.481510361]: Unable to solve the planning problem
[ INFO] [1617220857.579211583]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[ INFO] [1617220857.579620064]: Planning attempt 1 of at most 1
[ INFO] [1617220857.581726178]: Planner configuration 'arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1617220857.582119288]: RRTConnect: Starting planning with 1 states already in datastructure
[ERROR] [1617220857.914731988]: RRTConnect: Unable to sample any valid states for goal tree

## Version

ROS distribution :melodic 

Branch and commit you are using : master

## Steps to reproduce
Run the program and plan is rejected

1. *step 1...*
2. *step 2...*

## Code example (if necessary)

```
    # paste code here #
```

## Expected behavior

Describe the expected behavior (in regard to the bug described above).

## Any other information

Any other information you believe the maintainers need to know.

felixmaisonneuve, 2021-05-31:
Hi @mpatalberta, 
I have tried motion_plan.cpp on ROS Melodic and it seems to work fine. I didn't get an output like yours.
Were you trying with a real arm or were you using the MoveIt virtual robot?

Also, can you provide more information on what you did to get those errors?
What are the commands you did?

felixmaisonneuve, 2021-07-26:
Hi @mpatalberta ,

I will close this issue. If you still have a problem, feel free to re-open or create a new issue.

Thank you,
Felix

closed 2021-07-26 by felixmaisonneuve