Kinovarobotics/kinova-ros #334: Fake Controller and joint_state_publisher Publish Non-continuous Values in Moveit! Demo with virtual robot arm

engineering-coder, 2021-04-03:
## Description

I am trying to simulate a robot arm in Unity, and the idea is to get real-time robot joint angles and make Unity execute them simultaneously. However, when running the moveit demo with virtual robot arm `roslaunch j2s7s300_moveit_config j2s7s300_virtual_robot_demo.launch`, the fake controller of moveit and robot state publisher will both publish non-continuous joint angles, e.g. a certain joint angle suddenly changing from 4.71 rad to -1.53 rad (minus 2*pi) between two consecutive time frames, in topics `/joint_states` and `/move_group/fake_controller_joint_states`. This phenomenon happens when the execution of a planned trajectory begins. This non-continuity will cause unexpected behaviours in Unity and even cause the simulation to fail.

I wonder why is this happening and how can I get rid of it? I have thought about rounding all the values to -pi to pi (or any other proper ranges) before passing them to Unity, but will this cause other problems (e.g. when joint angle normally crosses pi, the rounded value will suddenly change from pi to some value near -pi)? 

## Version

ROS distribution: Kinetic

Branch and commit you are using: Not quite sure, maybe 1.2.1

felixmaisonneuve, 2021-05-31:
Hi @engineering-coder,
Kinova currently only supports robot simulation in Gazebo and RViz. If you want to simulate in Unity, you might have to change robot behaviour in your own fork of the repo.  
I was not able make a joint angle change from 4.71 rad to -1.53 rad like you observed, so I am not sure what causes this. 
To answer why this is happening, I know that the arm is limited in the angles it can go (e.g. it cannot do 2 full spins on itself), but I do not know why you are getting these values.

felixmaisonneuve, 2021-07-26:
Hi @engineering-coder ,

I will close this issue. If you still have a problem, feel free to re-open or create a new issue.

Thank you,
Felix

closed 2021-07-26 by felixmaisonneuve