Kinovarobotics/kinova-ros #335: actionlib.SimpleActionClient.wait_for_result() does not work

Shunichi09, 2021-05-18:
## Description

Referring to https://github.com/Kinovarobotics/kinova-ros/blob/master/kinova_demo/nodes/kinova_demo/pose_action_client.py, I implemented simple cartesian_pose_clients but I noticed that the wait_for_result() always returned  false although, the robot moved correctly.

## Version

ROS distribution : melodic

Branch and commit you are using : master

## Steps to reproduce

```sh
$ roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=j2s7s300
$ rosrun kinova_jaco2_arm_control bug_report_controller.py
```

## Code example (if necessary)

```my_controller.py
#! /usr/bin/env python
"""A set of example functions that can be used to control the arm"""
import rospy
import actionlib
import numpy as np

import kinova_msgs.msg
import geometry_msgs.msg
import tf
import std_msgs.msg
import math
import thread
from kinova_msgs.srv import *
from sensor_msgs.msg import JointState
import argparse

def cartesian_pose_client(position, orientation, prefix):
    """Send a cartesian goal to the action server."""
    action_address = '/' + prefix + 'driver/pose_action/tool_pose'
    client = actionlib.SimpleActionClient(action_address, kinova_msgs.msg.ArmPoseAction)
    client.wait_for_server()

    goal = kinova_msgs.msg.ArmPoseGoal()
    goal.pose.header = std_msgs.msg.Header(frame_id=(prefix + 'link_base'))
    goal.pose.pose.position = geometry_msgs.msg.Point(
        x=position[0], y=position[1], z=position[2])
    goal.pose.pose.orientation = geometry_msgs.msg.Quaternion(
        x=orientation[0], y=orientation[1], z=orientation[2], w=orientation[3])

    print('goal.pose in client 1: {}'.format(goal.pose.pose)) # debug

    client.send_goal(goal)

    if client.wait_for_result(rospy.Duration(10.0)):
        return client.get_result()
    else:
        client.cancel_all_goals()
        print('        the cartesian action timed-out')
        return None

def move():
    rospy.init_node("bug_report")
    
    HOME_TVEC = np.array([0.0, -0.4, 0.2])
    HOME_QUAT = np.array([np.sin(np.deg2rad(180.)/2.), 0., 0., np.cos(np.deg2rad(180.)/2.)]) 

    cartesian_pose_client(HOME_TVEC, HOME_QUAT, "j2s7s300_")

if __name__ == "__main__":
    move()
```

## Expected behavior

Should return True.

## Any other information

command line results.

``` 
[DEBUG] [1621335821.134538668]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): arm_comm_.isCloseToOther
[DEBUG] [1621335821.134608517]: Setting the current goal as succeeded
[DEBUG] [1621335821.134697265]: Setting status to succeeded on goal, id: /bug_report-1-1621335818.989, stamp: 1621335818.99
[DEBUG] [1621335821.134768813]: Publishing result for goal with id: /bug_report-1-1621335818.989 and stamp: 1621335818.99
[ WARN] [1621335821.134875525]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): LINE 176, setSucceeded 
[DEBUG] [1621335828.990191669]: The action server has received a new cancel request
[DEBUG] [1621335828.990315945]: Transitioning to a cancel requested state on goal id: /bug_report-1-1621335818.989, stamp: 1621335818.99
```

felixmaisonneuve, 2021-05-28:
Hi, 

So I copied your code in a file and tried it with a 6DOF with a spherical wrist (j2n6s300). 
The only thing I changed was the prefix in the `move()` function
```
cartesian_pose_client(HOME_TVEC, HOME_QUAT, "j2n6s300_")
```
and some prints in the condition 
```
if client.wait_for_result(rospy.Duration(10.0)):
        print("wait_for_result returned true")
        return client.get_result()
    else:
        print("wait_for_result returned false")
        client.cancel_all_goals()
        print('        the cartesian action timed-out')
        return None
```

Everything worked as expected
rosrun command output:
```
$ rosrun kinova_issues 335_bug_report_controller.py 
goal.pose in client 1: position: 
  x: 0.0
  y: -0.4
  z: 0.2
orientation: 
  x: 1.0
  y: 0.0
  z: 0.0
  w: 6.12323399574e-17
wait_for_result returned true
```

and my command line result : 
```
[DEBUG] [1622230791.256148694]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): arm_comm_.isCloseToOther
[DEBUG] [1622230791.256207226]: Setting the current goal as succeeded
[DEBUG] [1622230791.256295462]: Setting status to succeeded on goal, id: /bug_report-1-1622230788.838, stamp: 1622230788.84
[DEBUG] [1622230791.256418462]: Publishing result for goal with id: /bug_report-1-1622230788.838 and stamp: 1622230788.84
[ WARN] [1622230791.256778871]: void kinova::KinovaPoseActionServer::actionCallback(const ArmPoseGoalConstPtr&): LINE 176, setSucceeded
```

I do not get the last 2 debug message like in your result. 

Do you have other details that might help me reproduce your error? 
I will try it soon with the same robot configuration, as it seems it is the only difference between our executions

Shunichi09, 2021-05-30:
Thank you for your reply...
I'm really sorry to take your time.
I noticed that this error was caused by the initialization of the kinova jaco2.
(I didn't know that we should change the mode by the joystick after we turn on the robot.)
Anyway, I solved this problem and I really appreciate your reply.

closed 2021-05-30 by Shunichi09