Kinovarobotics/kinova-ros #336: Mico will not switch to torque control mode

chrismorley498, 2021-05-31:
## Description

The mico in our lab (m1n6s300) will not enter torque mode using the API and will also not respond to torque command through the TorqueConsole application in the SDK. I have tried all of the debugging on related issues but still have no luck. The Mico is set into a candle configuration, pointing straight up, when I attempt to switch into torque control mode. What is interesting to point out is that the mico responds to angular joint commands through the API and the DevelopementCenter. One thing I noticed is that when I launch the TorqueControl application through a call in the termianl, I get the following printed in my terminal.

chris@chris:~/Desktop/MICO2SDK/GUI$ ./TorqueConsole.sh
QMetaObject::connectSlotsByName: No matching signal for on_VibrationControllerActive_clicked()
QMetaObject::connectSlotsByName: No matching signal for on_VibrationControllerNotActive_clicked()
QMetaObject::connectSlotsByName: No matching signal for on_Home_pressed()
QMetaObject::connectSlotsByName: No matching signal for on_SetAngular_pressed()
QMetaObject::connectSlotsByName: No matching signal for on_SetCartesian_pressed()
QMetaObject::connectSlotsByName: No matching signal for on_SendActuatorGainDamping_pressed()
QMetaObject::connectSlotsByName: No matching signal for on_groupBox_3_toggled(bool)
QMetaObject::connectSlotsByName: No matching signal for on_VibrationControllerActive_pressed()
QMetaObject::connectSlotsByName: No matching signal for on_VibrationControllerNotActive_pressed()

Could this be related to the root problem? 

The only other thing I could think which is causing my problem is that my robot appears to be in assitive mode, based on the following line in the terminal where I run the kinova_robot.launch file:

[ INFO] [1622494729.436685537]: Found 1 device(s), using device at index 0 (model: Mico Assistive, serial number: , code version: 393736, code revision: 0)

This comment in [Issue #129](https://github.com/Kinovarobotics/kinova-ros/issues/129#issuecomment-426447867) mentions that the robot should be in service mode for torque control to work. If service mode is needed, how do I go about switching? I haven't been able to find any documentation on how to do so.

## Version

ROS distribution : Ubuntu 18.04

Branch and commit you are using : Using the Master branch, cloned to my catkin_ws two weeks ago

chrismorley498, 2021-06-01:
Following up, I found the executable in the SDK to active assistive mode. I was able to get the Mico into assistive mode by running the executable but that did solve solve my issue.

I've also added custom ROS_INFO calls to the KinComm functions to help debug. An example can be seen below (Final else{} body is what I added).

void KinovaComm::SetTorqueControlState(int state)
{
    int result;
    if (state)
    {
        //ROS_INFO("Switching to torque control");
        result = kinova_api_.switchTrajectoryTorque(TORQUE);
    }
    else
    {
        ROS_INFO("Switching to position control");
        result = kinova_api_.switchTrajectoryTorque(POSITION);
    }
    if (result != NO_ERROR_KINOVA)
    {
        throw KinovaCommException("Could not set the torque control state", result);
    }
    else
    {
      ROS_INFO("[custom message]: Torque Control Has been engaged");
    }
}

My custom message prints to the terminal which makes me think that everything should have worked properly? One obervation I made was that sometimes I get the following error message in my terminal when running a custom service I made which runs a rountine I made for switching into torque control mode.

[ERROR] [1622552417.443519069]: Exception thrown while processing service call: KinovaCommException: Could not set torque safety factor (return code: 2004)

I understand that the error code:2004 means there is a communication issue but this only occours ocassionally. Shouldn't the Mico be switching to torque control during the times where there is no error message?

felixmaisonneuve, 2021-06-01:
Hi @chrismorley498,
In your first comment you say your arm was already in assistive mode. Didn't you want to change it in service mode?

In TorqueConsole, can you try to : 
1. Click on "Test Connection" button. It should change to "Good Connection"
2. Put safety factor to '1'
3. Click on "Switch Torque Control" button. This should change your arm to torque control mode. (make sure you hold your arm when you do so)

After that you can click "Switch Trajectory Control" if you dont want to hold it anymore.

chrismorley498, 2021-06-01:
Hi @felixmaisonneuve , 

That was a typo, I meant to say I switched into service mode. No luck with either of the modes.

I've done those steps in the TorqueConsole and the robot still does not respond to any input from within the GUI. The test connection says "Good Connection". I then press the "Switch Control Torque" button. After doing that I try and use the other sliders and buttons to send a torque command to the joints. I get no reaction from the Mico for any of the joints.

felixmaisonneuve, 2021-06-01:
After a couple try on my arm (j2n6s300), I find the Torque Control to be very inconsistent. Looking at the issue you mentionned, there seems to have various errors that may cause this behaviour. 
I will try to identify what is the source of all of this and, with all the information you provided, I will try to fix your specific problem. 

I am currently working on it, I will keep you updated if anything comes up

felixmaisonneuve, 2021-06-01:
Ok @chrismorley498,
What you should try is : 
1. Put your arm in candle-like position (every joint in a vertical position)
2. Open Development Center (JacoSDK). In the "Monitoring" section, every actuator should have a value close to 180° in the Command and Position columns. Every actuator should also have a Torque value between -10 N*m and 10 N*m.
3. Open "Advanced Settings". In the section "Torque zero", select "apply to all". The "Monitoring" section should now show Torque values close to 0 for each actuator. 
4.  Open TorqueConsole. 
5. Test your connection with the "Test Connection" button. It should change to "Good Connection"
6. Change the "Safety Factor" to 1
7. Click "Switch Torque Control" button.
Your arm should now be in torque control mode

Tell me if this doesn't work

Regards,
Felix

chrismorley498, 2021-06-02:
Hi @felixmaisonneuve ,

I created a new catkin worspace and was able to get the Mico working with torque control through the API last night. However this morning it is refusing to switch back into torque control. It seems to be very intermitent, even last night I would need to reboot the Mico often as torque control would be dropped. I had tried those steps in the past with no luck although this morning I launched the TorqueConsole through the terminal as opposed to double-clicking it and torque control now works through there. I'm going to try and redo everything I did yesterday and see if I can get the API working again. Have you had any other customers with intermitent API torque control functionality? And if so was there a common solution?

chrismorley498, 2021-06-02:
One thing that I have noticed is that when I zero the torques through the development center, they all show as near zero in the monitoring section however upon immediately closing the development center, launching the kinova_robot.launch file and then echoing the m1n6s300_driver/out/joint_torques topics without touching the robot, The torques are no longer zero. Joints 2 and three have torque values just under 2Nm. Could this being the reason why the Mico is refusing to switch?

I call the setZeroTorque() function in my routine before calling the setTorqueControlState() function so I wonder if the function is not working properly for whatever reason although I checked and all the actuator adresses in that function match the addresses shown in the monitoring section of the development center. The only difference between the setZeroTorque() addresses and the monitoring section addresses is that the setZeroTorque() function has a 7th entry in the address vector which is 25 whereas only 6 actuators are listed (as expected) in the monitoring section. The line of code from the setZeroTorque() function can be seen below.

int actuator_address[] = {16,17,18,19,20,21,25};

chrismorley498, 2021-06-02:
Thought I would share the common error message I get when the connection drops in case that is helpful:

[m1n6s300_driver-2] process has died [pid 13558, exit code -6, cmd /home/chris/catkin_ws1/devel/lib/kinova_driver/kinova_arm_driver m1n6s300 __name:=m1n6s300_driver __log:=/home/chris/.ros/log/3123fe32-c3a6-11eb-997c-40167eb166cf/m1n6s300_driver-2.log].
log file: /home/chris/.ros/log/3123fe32-c3a6-11eb-997c-40167eb166cf/m1n6s300_driver-2*.log

felixmaisonneuve, 2021-06-02:
Ok, 
you say you got it working but it was intermitent through. This might be because of the Safety Factor. This factor is between 0 and 1. By default it is probably around 0.5 (not sure but this is not relevant). This factor will stop Torque Control State and bring back the robot to Cartesian Mode if the arm tries to move faster than `factor*MAX_ROBOT_SPEED` (or something like that). Since you were using the API, the Safety Factor was probably not set to 1. This means that, while moving the arm, it might have tried to move to fast and stopped abruptly. Which might explain what you saw. 

The Torque Control Mode is tricky, the force on the joints (the torque values) **have** to be close to 0, otherwise it might be a security issue and the robot won't change mode. This is why you need to setZeroTorque before changing mode. You say by echoing Torque_Joint topic you got values close to 2Nm, this is probably why Torque Control Mode is not working. Now, this doesn't explain why they are close to 2Nm if you say you setZeroTorque juste before and they appear close to 0 in Development Center. I don't have an answer to that and I don't know how this happened. 

I don't think this was the case, but make sure there is no terminal running a instance of `roslaunch kinova_bringup kinova_robot.launch robot_type="whatever"` **and** Development Center (or Torque Console) at the same time.  Otherwise, nothing will be working properly.

In your routine, when you call `setZeroTorque()`, is your robot in a candle-like position. This is not required but recommanded to reduce the effect of gravity on the torque sensors. If your robot is in a horizontal position, gravity could apply cross-torque force on the sensors. This could results in inconsistent measurements and unknow behaviour. 

To answer your questions :  
**Have you had any other customers with intermitent API torque control functionality?** yes and no
From what I know, intermitent functionnality is often caused by the torque sensors that has to be set to 0 before changing to Torque Control Mode. People will often assume you should be able to change mode whenever you want (without having to setZeroTorque), which is not the case since it could be a risk hazard. Also, this is a specific problem and not all users need this feature. I am also aware that there is probably a comminucation issue between ROS API and the arm controller. There is an error that pops up when running the `gravity_compensated_mode.py`. I am working on that. I think this communication error might also be related to your problem.
**And if so was there a common solution?** not really
Most times setting zero torque before changing mode is enough, but otherwise it is case by case. 
**Could this being the reason why the Mico is refusing to switch?** Yes
The arm, like I mentionned, will not change mode if the torque values are not close to 0

chrismorley498, 2021-06-02:
Hi @felixmaisonneuve 

My script sets the safety factor to 1 before calling torque control. I don't have conflicitng terminals open at the same time. My routine moves the Mico to the candle position, then does the following in order: zeros the torque, sets max/min torque, sets safety factor, sets the gravity vector, sets the maximum actuator velocity speed and then finally calls SetTorqueControlState().

Is it possible some of the torque sensors may need repair or is it more likely a software issue becuase they are zeroing properly in the monitoring section?

I've noticed sometimes when monitor the joint torques from m1n6s300/out/joint_torques, while the Mico is motionless, that there can be weird torque mesurements that dont makes sense (ie. negative torques where it must be positive at that joint in order for the robot to remain motionless). If I l tap the robot with a light-moderate amount of force the sensor readings will usually correct themselves.

felixmaisonneuve, 2021-06-02:
You are right, this is weird. It seems you have either a joint defect (I doubt it), there is comminucation problems related to torque or there is problems in the torque readings. 
You said the functionnality was interminent and would sometimes works and sometimes it wouldn't. Have you tried to tap your robot when it is working to see if it would "reset" torque readings and allow you to set Torque Control Mode.

Also you have mentionned the log file: /home/chris/.ros/log/3123fe32-c3a6-11eb-997c-40167eb166cf/m1n6s300_driver-2*.log.
Have you looked at it to see if there was something that could help us?

chrismorley498, 2021-06-02:
Yes I've tried taping it, doesn't seem to fix the problem. Just now I had the Mico running on torque control mode and it switched back to positional control while the mico was stationary, and had been for a few minutes. Another observation I made was that the optimal gravity parameters, calculated using runCOMParameterEstimation(robotType) were noticebly different today when compared to the ones I got yesterday. Using yesterdays parameters the robot will not support it's own weight but using the ones from today it will (when torque mode works that is). 

Despite the fact that error message mentions that log file, it does not appear that the file was actually created as there is nothing at that file location. I'll keep checking to see if I can find it elsewhere but after quickly checking online I see that others have had those files not save as well.

felixmaisonneuve, 2021-06-02:
Ok, in your kinova_arm.cpp can you add `return true` in setTorqueControlModeService like so : 
```
bool KinovaArm::setTorqueControlModeService(kinova_msgs::SetTorqueControlMode::Request &req, kinova_msgs::SetTorqueControlMode::Response &res)
{
    kinova_comm_.SetTorqueControlState(req.state);
    return true;
}
```
You will have to catkin_make your workspace again.
This should get rid of the occasional and give a more consistent behaviour
```
[ERROR] [1622552417.443519069]: Exception thrown while processing service call: KinovaCommException: Could not set torque safety factor (return code: 2004)
```

After that, can you try again to see if you can run your script or if it still fails. If it does, can you write your exact steps and any Warning/error/information that are relevant.

Also, if it fails, can you try to run the gravity_compensated_mode.py provided in kinova_demo `rosrun kinova_demo gravity_compensated_mode.py m1n6s300`
If this script works it probably means the errors are coming from your script.

I said the error was probably software related (communication error) and not mecanical (bad readings or joint defect), but if this doesn't work, I am not sure anymore. 

I have tried a couple of things and I cannot get any errors consistantly. I can change to Torque Control Mode basically everytime.

chrismorley498, 2021-06-02:
I added the "return true;" at the end of that function. I haven't seen the associated error flag that that function was tripping so I think that must have fixed that issue. This afternoon I have managed to get the robot into torque control for a good amount of time however I am still constantly needing to reboot the Mico between these times where is works. After adding the "return true;" I've tried to switch into torque control mode 4 times, 2 of which worked and the other two failed. Below you can see the terminal output from the terminal where I call kinova_robot.launch. The output contains custom messages I have added in both my service and other functions to help for debugging purposes.

[ INFO] [1622665031.492145155]: //////////////////////////////////////////////////////////////////////
[ INFO] [1622665031.492194850]: [custom message]: The torque service server has been called
[ INFO] [1622665032.504360009]: Starting in 10
[ INFO] [1622665033.504633215]: Starting in 9
[ INFO] [1622665034.504869841]: Starting in 8
[ INFO] [1622665035.505064744]: Starting in 7
[ INFO] [1622665036.505311452]: Starting in 6
[ INFO] [1622665037.505570339]: Starting in 5
[ INFO] [1622665038.505837206]: Starting in 4
[ INFO] [1622665039.506098205]: Starting in 3
[ INFO] [1622665040.506332679]: Starting in 2
[ INFO] [1622665041.506599651]: Starting in 1
[ INFO] [1622665041.509184491]: Setting the gravity vector
[ INFO] [1622665042.521137357]: [custom message]: Torques Measurements have been zeroed
[ INFO] [1622665043.521400284]: Setting min torues - -37.000000 -37.000000 -37.000000 -8.000000 -8.000000 -8.000000 -8.000000
[ INFO] [1622665043.521512528]: Setting max torues - 37.000000 37.000000 37.000000 8.000000 8.000000 8.000000 8.000000
[ INFO] [1622665043.523153928]: [custom message]: There was no error flag setting the min/max torque with the kinova_api_ call
[ INFO] [1622665043.525121523]: [custom message]: The KinovaComm & kinova_api_ didn't have any error flags when setting the torque control FOS
terminate called after throwing an instance of 'kinova::KinovaCommException'
  what():  KinovaCommException: Could not get the angular command (return code: 1015)

[m1n6s300_driver-2] process has died [pid 5684, exit code -6, cmd /home/chris/catkin_ws1/devel/lib/kinova_driver/kinova_arm_driver m1n6s300 __name:=m1n6s300_driver __log:=/home/chris/.ros/log/7fc7efc4-c3df-11eb-a65a-40167eb166cf/m1n6s300_driver-2.log].
log file: /home/chris/.ros/log/7fc7efc4-c3df-11eb-a65a-40167eb166cf/m1n6s300_driver-2*.log

The steps leading up to the countdown you see in my terminal output error are as follows: 

1. call kinova_robot.launch (Everything seems fine here, terminal is populated with green text) Last line is:[DEBUG] [1622666415.553003554]: Trying to publish message of type [actionlib_msgs/GoalStatusArray/8b2b82f13216d0a8ea88bd3af735e619] on a publisher with type [actionlib_msgs/GoalStatusArray/8b2b82f13216d0a8ea88bd3af735e619]

2. In a new terminal call my custom service named "engageTorqueControl". This service first uses the API to move to the candle position (This part works most of the time, if it doesn't a quick reboot fixes it)
3. I then set the gravity vector to {0,0,-9.81}, zero the torque measurements, set min/max torque values, set safety factor and then finally set control to torque mode.

The countdown you see if there so that the Mico has time to arrive at the candle position from any other configuration before another function is called (I found this improved things quite a bit). I also have a 1 second delay between all other functions in the engageTorqueControl service.

At this point I am able to do some work with the Mico in torque control but just constantly needing to reboot it. It seems as though it takes a couple hours each morning constantly trying before it will make the initial switch. After making the intial switch it is then intermitent as I described. It is also strange that these errors only occour when calling torque related commands. 

gravity_compendated_mode.py is working all of a sudden (maybe this has to do with creating a new catkin workspace). 

I'm almost inclined to think that the API is getting incorrect joint torque measurements for joints 2 & 3. This could explain why my COM parameters were noticebly different today when compared to yesterday. Just strange that this only seems to be happening for joints 2 and 3. 

Also just wanted to mention I really appreciate all your help debugging this, very strange issue.

felixmaisonneuve, 2021-06-03:
If that `return true` seems to fix some problems, that is a good start.
The `return code: 1015` is a ERROR_NO_DEVICE_FOUND. I found the issues #218 and #171 that are related to this. They might be helpfull, but I think they were not fully answered.

I will contact a colleague to see if this might be caused by an hardware problem.

I won't be able to help you more today,  but I will be back on this tomorrow.

It is a pleasure to help you, it is a strange issue

chrismorley498, 2021-06-03:
I'll check those issues out and see if I get anywhere with those. I'll be out of the lab tomorrow, back Monday so no rush. Thanks again!

felixmaisonneuve, 2021-06-04:
I don't know if those issues were of any help. 

I talked to some people and it turns out that, in some cases, the `setZeroTorque()` function does not reset properly the torque sensors. What you should do then is to reset the torque sensor (while in candle position) using : 
Development Center -> Advanced Settings -> Torque Zero -> Apply to al
We will see if this seems to improve the behaviour of you program when you run it afterwards. 

I am not sure why you get `return code: 1015` errors apparently out of nowhere. 
You could send your program and I could try it on my arm to see if it also crashes on it or if it is something related to your arm.

chrismorley498, 2021-06-08:
Sorry for the delay in getting back to you. I went through and tried some of the suggestions in the other issues you linked, I was very successful Thursday switching into torque control mode so I'm inclined to think that some of the issues were common there. For anyone else having the same issue I think that it was the suggestion in rule #104 which helped solve part of my issue.  The only time I seemed to get errors after making those adjustments were when I tried to run my custom initiate torque control routine after I had already called that rountine on the same power cycle of the Mico. I made another routine to switch back to position control which I call prior to switching to torque control (if torque control has already been engaged) and that seemed to fix all the issues I was having. I think there was an issue with my routine that only became apparent when It was called multiple times on the same power cycle. 

@felixmaisonneuve Thanks alot for all your help! This resolves the issue switching into torque control and If I have other errors it would be best I open an new issue with specifics to any future error.

felixmaisonneuve, 2021-06-08:
Hi @chrismorley498,
That is great news.

I will now close this issue

Best,
Felix

closed 2021-06-08 by felixmaisonneuve