Kinovarobotics/kinova-ros #339: Destruction entered while using Kinova driver

davidDS96, 2021-06-29:
Dear community,

I am using the latest version of the Kinova-ROS stack together with ROS Melodic to control a Kinova Jaco js2s6s200. However, when launching the driver using the bringup-launch file, a DEBUG message gets thrown (as displayed in the picture below). After a while, an ERROR message displaying _DESTRUCTION_ENTERED_ (as displayed in the picture below). 

It is possible to control the robot through the demo code (_joints_action_client.py_) once, but after controlling it the first time, it is no longer possible to let the robot move.

Does anyone know what the issue could be?

David

[image]

felixmaisonneuve, 2021-06-29:
Hi @davidDS96
The debug messages are normal, so nothing to worry about there.

Have you made any changes in the kinova-ros package? You say this error shows up after a while. Do you do a specific sequence to trigger it? Does it trigger even if you just let the arm on "idle" after running `roslaunch kinova_bringup kinova_robot.launch`?

Can you make sure your system is fully updated with `sudo apt update && sudo apt upgrade`

Best,
Felix

davidDS96, 2021-06-29:
Hi @felixmaisonneuve 
I didn't make any changes in the kinova-ros package. I just cloned the master branch and used it as is. I don't know what triggers the event. Sometimes it happens after starting up or while the robot is idle. Sometimes it takes 4 hours to show up after moving the robot normally (using MoveIt! for instance).

My system is fully updated and still the error messages in red are displayed.

Kind regards,
David

felixmaisonneuve, 2021-06-29:
Ok, can you do a `rostopic list` and copy it here after launching the driver to see if the `/j2s6s200_driver/trjectory_controller/state` does exist?

davidDS96, 2021-06-29:
Hi @felixmaisonneuve 

In the picture below, you can find a screenshot of a terminal running the bringup.launch (left terminal) and a terminal showing the output of ```rostopic echo /j2s6s200_driver/trajectory_controller/state```. The topic ```/j2s6s200_driver/trajectory_controller/state``` is present.

Currently, we suspect the issue to be a combination of a 25 meter long Ethernet cable and a switch causing communication problems with the Kinova driver. Could this be the source of our issue?

[image]

felixmaisonneuve, 2021-06-29:
This error seems very inconsistent, so it is worth trying by using another cable without a switch. It will allow us to see if this comes directly from your computer or not.

felixmaisonneuve, 2021-06-29:
This is good news. 
I hope that switch isn't mandatory in your setup. If it is, you could try with another one. That particular switch might have a defect.

Since your problem seems to be fixed and it is not related to our driver (at first sight), I will close this issue. 
If, for whatever reason, this error comes up again, feel free to reopen this issue.  

Regards,
Felix

IvoD1998, 2021-07-01:
@davidDS96 and I have been looking into the issue further, but the problem keeps occuring when using an ethernet connection and a switch together. We used wireshark to look at what is being transmitted and received and noticed the problem takes place when the pc is sending trajectory data to the robot (the robot being IP 10.42.0.69) as displayed in the image below. 
As the problem can occur immediately, as well as after hours of using the robot, we are still unsure if this actually is an issue caused by the switch/cable or a driver related problem @felixmaisonneuve.

Regards
Ivo 

[image]

felixmaisonneuve, 2021-07-05:
Hi @IvoD1998,

Have you tried using different switches? It seems odd to me a switch would modify the data that is being transmitted. 
Are we sure this never occurs when there is no switch involved?

IvoD1998, 2021-07-05:
hi @felixmaisonneuve 
We have tried using different switches, which all result in the same issue. However, currently we are trying to use the robot without using any switch and have not encountered the problem thus far. If the problem does occur without the switch, I will let you know.

Regards
Ivo

felixmaisonneuve, 2021-07-05:
If the problem only appear when using a switch, could it be possible there is interference within the different devices connected to the switch (I don't really know how ethernet communication/switches work)? 

Maybe a wrong network configuration?

I will leave this issue open for now. It will let you some time to do further testing to identify where this is coming from. 

Regards,
Felix

felixmaisonneuve, 2021-07-27:
Hi @davidDS96 and @IvoD1998,

Did you do some more testing? 

Having a long cable (you mentionned a 25m long cable) can result in packet loss. This is because the voltage will be lower at the edges when using a longer cable. This could result in bad/undefined readings. However, from what I read, it shouldn't be a problem for a cable of less than 100m, so you should be fine. 

Maybe what you are experiencing is due to packet loss because of the (somewhat) long cable and the switch (but I doubt it) or something related (e.g. thin or damaged cable, defective switch). 

Regards, 
Felix

IvoD1998, 2021-07-28:
Hi @felixmaisonneuve 

We did some more testing and found that the issue is gone when we don't use a switch, but communicate with the robot directly over ethernet.

regards
Ivo

closed 2021-07-28 by felixmaisonneuve

davidDS96, 2022-06-23:
Hi @JoseugeLABS,

We weren't able to make the Kinova stack work with an ethernet connection, nor directly or using a switch between the PC and the Kinova controller. The same issue arises occasionally. 

Now, we switched to the USB connection and we didn't experience any connection issues so far.

Kind regards,
David