Kinovarobotics/kinova-ros #345: Issues starting Cartesian Admittance mode

ericrosenbrown, 2021-07-28:
Hello,

We are trying to activate Cartesian Admittance mode for our Jaco arms. After launching the startup script (roslaunch movo_bringup movo_system.launch), we try to follow the instructions in the tutorial and want to run:

rosservice call /'${kinova_robotType}_driver'/in/start_force_control
rosservice call /'${kinova_robotType}_driver'/in/stop_force_control

However, there are no rosserives that have start_force_control in them (although there are many other rosserices activated). Do we need to run another script to get these rosservices activated, or is there something missing in our startup script? Otherwise, is there another way to get admittance mode running?

Thanks

felixmaisonneuve, 2021-07-28:
Hi @ericrosenbrown ,

I did some testing and I am pretty sure `start_force_control` and `stop_force_control` are not doing anything. You need to use `/j2n6s300_driver/in/set_torque_control_mode "state: 1"` to activate the admittance and `/j2n6s300_driver/in/set_torque_control_mode "state: 0"` to stop it.

But before calling the service, you need to make sure the robot is in a candle-like position (in a vertical position), and it is possible you need to reset torques with `set_zero_torque` service. Torque values can change over time and if they are above a certain value, the robot won't activate the admittance mode. 

Let me know how it goes,
Felix

felixmaisonneuve, 2021-08-20:
Hi @ericrosenbrown,

I will assume this has fixed your problem and I will close this issue.
If you still encounter the problem, comment / re-open it or create a new issue.

Regards,
Felix

closed 2021-08-20 by felixmaisonneuve