Kinovarobotics/kinova-ros #350: Slow movement of arm using velocity control, and unexpected movement using joint position control

phiggin1, 2021-08-19:
## Description

I was trying to be a Jaco arm controllable using MoveIT, but it results in very slow movement and the arm failing to reach the goal. It looks like when given joint velocity commands the arm does not match the provided velocities. While looking into joint position and cartesian position control the arm works as expected when using cartesian control, but when using joint position control the arm ends up moving more joints than just the ones that were commanded.

## Version

ROS distribution : 

Branch and commit you are using: melodic-devel, 9fcfe97

## Steps to reproduce

roslaunch kinova_bringup kinova_robot.launch kinova_robot_type:=j2n6s300
roslaunch j2n6s300_moveit_config j2n6s300_demo.launch
Then in rviz after choosing a random valid goal pose, selecting plan and execute the robot moves very slowly toward the goal but stops before reaching it.

When directly giving a joint velocity command, the reported velocity is much lower than what was commanded.
rostopic pub -r 100 /j2n6s300_driver/in/joint_velocity kinova_msgs/JointVelocity "{joint1: 0.0, joint2: 0.0, joint3: 0.0, joint4: 0.0, joint5: 0.0, joint6: 10.0}"

rostopic echo /j2n6s300_driver/in/joint_velocity 
joint1: 0.0
joint2: 0.0
joint3: 0.0
joint4: 0.0
joint5: 0.0
joint6: 10.0
joint7: 0.0

rostopic echo /j2n6s300_driver/out/joint_state/velocity 
[4.4e-46, 4.4e-46, 4.4e-46, 4.4e-46, -4.4e-46, 0.040377741571453345, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]

rosrun kinova_demo pose_action_client.py -v -r j2n6s300 mdeg -- 0.01 0 0 0 0 10
Using the arm with cartesian position control works fine with the arm moving at what looks like the proper speed and getting to the goal position.

rosrun kinova_demo joints_action_client.py -v -r j2n6s300 degree -- 0 0 0 0 0 10
However, when controlling the arm with joint position control the arm moves more than just the joint that is commanded. When commanding joint 6 to rotate 10 degrees both shoulder joints also move.

rosrun kinova_driver kinova_interactive_control j2n6s300
Using the interactive controls and manipulating the interactive markers in rviz shows the same behavior. The cartesian control works as expected, but when using joint control joints beyond than the ones commanded move.

## Expected behavior

I would expect the behavior to be consist across all the different control methods, and for the joint velocities commanded and reported to match each other.

## Any other information

I believe that the arm had been working correctly prior to covid. After getting access to the arm after 18 months when it was first turned on it was stuck with a flashing green light on the controller. Reinstalling the firmware (to version 6.2.8) did fix the issue.

felixmaisonneuve, 2021-08-19:
Hi @phiggin1,

Thank you for using the provided template, it really helps solving the issue.

This is a very strange issue. The commit you are using is in the melodic-devel branch (like you mentionned). Can you confirm your are using ROS Melodic or ROS Kinetic? Also, what is your ubuntu version?

I tried with Ubuntu 18 on ROS Melodic on the specific commit you mentionned and did not experienced anything like that. 
Did you make any modifications to the kinova-ros package?

Can you also provide the full output after you run the command `roslaunch kinova_bringup kinova_robot.launch kinova_robot_type:=j2n6s300`?

You say you experience slow movement when publishing on the joint speed topic, so I doubt this is directly related to MoveIt. Maybe some velocity limits when using joint_speed commands...

Regards,
Felix

phiggin1, 2021-08-19:
I am using ROS Melodic on Ubuntu 18.04.5 and there weren't  any modifications to the kinova-ros package, and here is output for the roslaunch kinova_bringup kinova_robot.launch kinova_robot_type:=j2n6s300:
... logging to /home/husky/.ros/log/b7675b14-0123-11ec-a8f4-003018c1985d/roslaunch-iral-husky-6624.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro.py is deprecated; please use xacro instead
started roslaunch server http://iral-husky:39159/

SUMMARY
========

PARAMETERS
 * /j2n6s300_driver/connection_type: USB
 * /j2n6s300_driver/ethernet/local_broadcast_port: 25025
 * /j2n6s300_driver/ethernet/local_cmd_port: 25000
 * /j2n6s300_driver/ethernet/local_machine_IP: 192.168.100.100
 * /j2n6s300_driver/ethernet/subnet_mask: 255.255.255.0
 * /j2n6s300_driver/jointSpeedLimitParameter1: 10
 * /j2n6s300_driver/jointSpeedLimitParameter2: 20
 * /j2n6s300_driver/robot_name: j2n6s300
 * /j2n6s300_driver/robot_type: j2n6s300
 * /j2n6s300_driver/serial_number: not_set
 * /j2n6s300_driver/status_interval_seconds: 0.1
 * /j2n6s300_driver/torque_parameters/publish_torque_with_gravity_compensation: False
 * /j2n6s300_driver/torque_parameters/use_estimated_COM_parameters: False
 * /j2n6s300_driver/use_jaco_v1_fingers: False
 * /kinova_number_of_robots: 2
 * /kinova_robots: [{'serial': 'PJ00...
 * /robot_description: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.11

NODES
  /
    j2n6s300_driver (kinova_driver/kinova_arm_driver)
    j2n6s300_state_publisher (robot_state_publisher/robot_state_publisher)

ROS_MASTER_URI=http://localhost:11311

process[j2n6s300_driver-1]: started with pid [6642]
process[j2n6s300_state_publisher-2]: started with pid [6643]
[ INFO] [1629401577.502914595]: kinova_robotType is j2n6s300.
[ INFO] [1629401577.503862554]: kinova_robotName is j2n6s300.
[ INFO] [1629401577.506721840]: Initializing Kinova USB API (header version: 50300, library version: 5.2.0)
[ INFO] [1629401577.667777045]: Found 1 device(s), using device at index 0 (model: Jaco V2 Rehab, serial number: , code version: 393736, code revision: 0)
[ INFO] [1629401577.738584077]: Initializing fingers...this will take a few seconds and the fingers should open completely
[ INFO] [1629401577.823948438]: The arm is ready to use.
j2n6s300_joint_1 j2n6s300_joint_2 j2n6s300_joint_3 j2n6s300_joint_4 j2n6s300_joint_5 j2n6s300_joint_6 
[DEBUG] [1629401577.855476721]: Trying to publish message of type [control_msgs/FollowJointTrajectoryFeedback/10817c60c2486ef6b33e97dcd87f4474] on a publisher with type [control_msgs/FollowJointTrajectoryFeedback/10817c60c2486ef6b33e97dcd87f4474]
[DEBUG] [1629401577.935665601]: Trying to publish message of type [kinova_msgs/JointAngles/bf16c4e5154848a0767d1aaab0a4796b] on a publisher with type [kinova_msgs/JointAngles/bf16c4e5154848a0767d1aaab0a4796b]
[DEBUG] [1629401577.935904421]: Trying to publish message of type [sensor_msgs/JointState/3066dcd76a6cfaef579bd0f34173e9fd] on a publisher with type [sensor_msgs/JointState/3066dcd76a6cfaef579bd0f34173e9fd]
[DEBUG] [1629401577.939764245]: Trying to publish message of type [geometry_msgs/PoseStamped/d3812c3cbc69362b77dc0b19b345f8f5] on a publisher with type [geometry_msgs/PoseStamped/d3812c3cbc69362b77dc0b19b345f8f5]
[DEBUG] [1629401577.939871512]: Trying to publish message of type [kinova_msgs/KinovaPose/e831d993faea563f6fe69d7db9b384c9] on a publisher with type [kinova_msgs/KinovaPose/e831d993faea563f6fe69d7db9b384c9]
[DEBUG] [1629401577.941590638]: Trying to publish message of type [geometry_msgs/WrenchStamped/d78d3cb249ce23087ade7e7d0c40cfa7] on a publisher with type [geometry_msgs/WrenchStamped/d78d3cb249ce23087ade7e7d0c40cfa7]
[DEBUG] [1629401577.943728947]: Trying to publish message of type [kinova_msgs/FingerPosition/f56891e5dcd1900989f764a9b845c8e5] on a publisher with type [kinova_msgs/FingerPosition/f56891e5dcd1900989f764a9b845c8e5]
[DEBUG] [1629401578.042679162]: Trying to publish message of type [actionlib_msgs/GoalStatusArray/8b2b82f13216d0a8ea88bd3af735e619] on a publisher with type [actionlib_msgs/GoalStatusArray/8b2b82f13216d0a8ea88bd3af735e619]

I have also just tried using the virtual joystick in the Kinova SDK and when doing angular control it looks like the arm should be moving a joint at 30 deg/s but was moving about as slow as what I was seeing in ROS, so it might be a software problem with the arm? This has been a weird one.

Thanks

felixmaisonneuve, 2021-08-20:
Hi @phiggin1,

Doing some experimentation, I realised there is velocity limits that only applies to angular movement and it would explain your behaviour (if they are incorrect). They seem to be defined in the base's firmware and I am not sure if you can modify these or define other limits. 

First thing to do will be to verify if your limits seem normal. To do that, open Development Center. In the *General settings* window, switch to angular control mode (there is a white button for that). After that, go to the *Monitoring* window and compare your velocity limitations with mine : 
[image]

If they are different, it is possibly a firmware issue and I would recommend to flash your base's firmware again and see if it seems to fix the issue.

If they are the same, I will investigate to see if it is possibly to set angular limits using ROS / Development Center.

Regards,
Felix

phiggin1, 2021-08-23:
When going into General settings window and switching to angular control mode the second joint immediately started to rotate the arm down from the home position. Reinstalling the firmware didn’t change the behavior.
Looking at the Monitoring window there were communication errors being reported. 
[image]

felixmaisonneuve, 2021-08-23:
At this point I don't think the issue is coming from ROS. I would suggest you to send an email to our support team support@kinova.ca. This is probably some kind of hardware issue. Just out of curiosity, what are your firmware version for your actuators?

You should have something similar to this (except for your firmware version which should be 6.2.8) :  
[image]

phiggin1, 2021-08-23:
Thanks for the help, that was my thinking as well, I will be contacting support. It looks like the actuator firmware is rather old as it is showing as version 1.5.1 in the dev kit.
[image]

felixmaisonneuve, 2021-08-23:
Yes, I would recomend contacting support since they seem quite old.

One last thing also, I see your arm is in Assistive mode. I previously saw (on old issues) that it is better to put your arm in Service mode. There is executable files in *JACO-SDK->RobotConfiguration* to switch between them. Maybe it could help.

I don't think I can help you more with this issue so I will close it, but feel free to comment or re-open it if anything comes up

Regards,
Felix

closed 2021-08-23 by felixmaisonneuve