Kinovarobotics/kinova-ros #353: ROS2 foxy and ubuntu 20.04

Betancourt20, 2021-09-09:
Hello
I'm gonna start the working with the Jaco and I already have installed ubuntu 20.04 in my computer. Before continue installing the ROS stuff, I was wondering if you have in mind to update the code for ros2 foxy? or How can I help to update the code  for this new version?
Best regards,
Julio

felixmaisonneuve, 2021-09-09:
Hi @Betancourt20,

We consider adding support to ROS2, but I don't think this will be done in the short term (I don't have a date to give you). We currently only support ROS. This means you would need to use ROS Noetic (at the moment). 

I have not looked in details at what needs to be done to move from ROS to ROS2, but, at first glance, I think this requires quite a bit of work. 
I would start by looking at the ROS2 [documention](https://docs.ros.org/en/foxy/Contributing/Migration-Guide.html). 

I will leave this issue open and keep you updated whenever there is progress regarding ROS2 migration. 
If you decide to do it yourself, you are free to leave comments/questions here, but I don't know if I can help that much. 

Regards,
Felix

skpawar1305, 2023-01-31:
interactive mode works now in humble