Kinovarobotics/kinova-ros #354: All topics are not listed

divishad12, 2021-09-10:
[image]

I have launched the physical robot by  'roslaunch kinova_bringup kinova_robot.launch' and when I saw the list of topics it accessed, there was only one OUT topic whereas there used to be number of topics beforehand last week. Please let me know if something is wrong with the robot or the ROS package?

felixmaisonneuve, 2021-09-10:
Hi @divishad12,

Is your branch up to date? Are you on the correct branch (melodic-devel or noetic-devel)?
Can you provide the full output of `roslaunch kinova_bringup kinova_robot.launch`?

Regards,
Felix

divishad12, 2021-09-12:
SUMMARY
========

PARAMETERS
 * /j2n6s300_driver/connection_type: USB
 * /j2n6s300_driver/ethernet/local_broadcast_port: 25025
 * /j2n6s300_driver/ethernet/local_cmd_port: 25000
 * /j2n6s300_driver/ethernet/local_machine_IP: 192.168.100.100
 * /j2n6s300_driver/ethernet/subnet_mask: 255.255.255.0
 * /j2n6s300_driver/jointSpeedLimitParameter1: 10
 * /j2n6s300_driver/jointSpeedLimitParameter2: 20
 * /j2n6s300_driver/robot_name: j2n6s300
 * /j2n6s300_driver/robot_type: j2n6s300
 * /j2n6s300_driver/serial_number: not_set
 * /j2n6s300_driver/status_interval_seconds: 0.1
 * /j2n6s300_driver/torque_parameters/publish_torque_with_gravity_compensation: False
 * /j2n6s300_driver/torque_parameters/use_estimated_COM_parameters: False
 * /j2n6s300_driver/use_jaco_v1_fingers: False
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.15.11

NODES
  /
    j2n6s300_driver (kinova_driver/kinova_arm_driver)
    j2n6s300_state_publisher (robot_state_publisher/robot_state_publisher)

auto-starting new master
process[master]: started with pid [10484]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to eda681d8-13c4-11ec-944a-bf6f49b85bd6
process[rosout-1]: started with pid [10494]
started core service [/rosout]
process[j2n6s300_driver-2]: started with pid [10501]
process[j2n6s300_state_publisher-3]: started with pid [10502]
[ INFO] [1631449707.734291602]: kinova_robotType is j2n6s300.
[ INFO] [1631449707.736553548]: kinova_robotName is j2n6s300.

This is what I am getting after roslaunch kinova_bringup kinova_robot.launch
And yes the branch is noetic and is updated. I don't know if something is wrong from my end or Robot's end. 

Thank You

felixmaisonneuve, 2021-09-13:
Hi @divishad12,

Your ROS output is incomplete and the initialization of the arm does not finish correctly.

I will need you to verify 2 things.

1. Is your arm really a j2n6s300?
2. Can you make sure your arm is correctly plugged in. Also can you launch *Development Center* to see if it detects the arm. 

Regards,
Felix

divishad12, 2021-09-13:
Hi @felixmaisonneuve ,

I checked both the things and 
    1. Yes it is the same arm
    2. Development Center is detecting the arm and I am able to move the arm using virtual joystick too.

Regards,
Divishad

felixmaisonneuve, 2021-09-13:
Did you modified any of the files in the kinova_driver folder?

Can you make sure you are using the correct branch (melodic-devel or noetic-devel)?
Maybe you could try to update and recompile your driver with `git pull` and `catkin_make`.

divishad12, 2021-09-13:
Hi @felixmaisonneuve ,

Looks like the workspace was corrupted due to inclusion of a new package. I tried removing those packages and adding again (incl. Kinova_ROS) and then sourcing the workspace again. Now it is working fine. Thank you for your assistance.

Regards,
Divishad

felixmaisonneuve, 2021-09-13:
No problem. I'm glad everything is working now.

Regards,
Felix

closed 2021-09-13 by felixmaisonneuve